ARISE-Initiative/robosuite #369: Cannot initialize a EGL device display(non-Nvidia graphics)

yuancheng-li, 2022-12-10:
error：
`~$ python -m robosuite.demos.demo_random_action
[robosuite WARNING] No private macro file found! (__init__.py:7)
[robosuite WARNING] It is recommended to use a private macro file (__init__.py:8)
[robosuite WARNING] To setup, run: python /home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/scripts/setup_macros.py (__init__.py:9)
Traceback (most recent call last):
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/__init__.py", line 11, in <module>
    from robosuite.environments.base import make
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/environments/base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/utils/binding_utils.py", line 58, in <module>
    from robosuite.renderers.context.egl_context import EGLGLContext as GLContext
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/renderers/context/egl_context.py", line 32, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 67, in <module>
    raise ImportError(
ImportError: Cannot initialize a EGL device display. This likely means that your EGL driver does not support the PLATFORM_DEVICE extension, which is required for creating a headless rendering context.
`
platform：Ubuntu 20.04，Inter nuc， Non-NVIDIA graphics card（My GPU is Intel Corporation UHD Graphics）
what should I do if I want to use this robosuite module

zhuyifengzju, 2023-03-26:
Hi,
     You would want to first double check if your graphics card can support `EGL`, and make sure `egl` is installed on your device. If `egl` is not possible, you need to use `osmesa` or `glfw`. You can take a look at to enable `glfw` or `osmesa` https://github.com/ARISE-Initiative/robosuite/blob/60b358673cc5aaccbf1dc950dbfe365e0915c69a/robosuite/utils/binding_utils.py#L37

CHbloom, 2023-03-30:
have you solved it?I met the same problem

zhuyifengzju, 2023-04-15:
Hi, do you have nvidia gpu installed on your end? And if not, have you tried with other gpu rendering options as I mentioned above?

aniri15, 2023-05-11:
> 

hi I also met this same error and I tried both installations. Do you have some recommadations regarding this? I also switched from egl to osmesa glfw and glx since I use a Intel graphic card.

warmmm, 2023-06-24:
I have the same problem, I think the main issue is the VMware can not use the GPU, so how could slove this problem?

zhuyifengzju, 2023-06-24:
Hi,

     If you are using virtual machines and cannot use GPU, the options can only be the CPU rendering. The robosuite rendering is basically a wrapper over the mujoco rendering. Does the mujoco rendering work all fine in your setting? If mujoco rendering works and robosuite rendering doesn't, we would look into where the mismatch comes from. If mujoco rendering doesn't work in the first place, I suggest opening an issue in the mujoco repo.

warmmm, 2023-06-25:
I have install Ubuntu in my disk, but it still have the same issues,the EGL is still failure initial, so how can I do? I can use the mujoco's offical models.

warmmm, 2023-06-26:
when I use the example code that you provide in Quick Start, it also happen the EGL error, like: 

Exception ignored in: <function EGLGLContext.__del__ at 0x7f1afe98f7f0>
Traceback (most recent call last):
  File "/home/chen/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
  File "/home/chen/robosuite/robosuite/renderers/context/egl_context.py", line 150, in free
  File "/home/chen/.local/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>

I don't know what problem of EGL?

zhuyifengzju, 2023-06-26:
My point is that you should first check if the EGL rendering in mujoco repo works fine on your machines. Based on the provided information so far, it seems to be a GL installation issue, not a bug from our codebase. 

One more important note is that, if your machine does not have GPUs, you will need to use non-gpu rendering osmesa instead of EGL. EGL is for headless rendering with GPUs.

warmmm, 2023-06-26:
I can run the file robosuite/demos/demo_random_action.py, but it qiut after 1 minite, and the bug is also Exception ignored in: <function MjRenderContext.__del__ at 0x7faf769505e0>
Traceback (most recent call last):
  File "/home/chen/robosuite/robosuite/utils/binding_utils.py", line 199, in __del__
  File "/home/chen/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/chen/anaconda3/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
Exception ignored in: <function EGLGLContext.__del__ at 0x7faf76950430>
Traceback (most recent call last):
  File "/home/chen/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
  File "/home/chen/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/chen/anaconda3/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>

But I have install the NVIDIA driver, so what's the problem?

zhuyifengzju, 2023-06-26:
If it can run for a minute and the errors you posted are the only things you have, that means there is no problem in running robosuite. I believe this is just that the example script didn't call `env.close()`. We will make a pull request for this example script to prevent this error message. As for normal usage, I don't think there is any issue with your installation unless you have other errors.

warmmm, 2023-06-30:
ok, thank you. I have no probelm now.