ARISE-Initiative/robosuite #370: OpenGL EGL Error

kxgao, 2022-12-13:
Hello, I am running robosuite 1.4 with mujoco 2.3.1 in headless rendering mode, on a GeForce RTX GPU. When running a script to generate a trajectory from some environment, I always get the error `OpenGL.raw.EGL._errors.EGLError: <exception str() failed>`. The trajectory is generated without issue.

To reproduce this error:
- Create Anaconda environment with Python 3.9
- pip install mujoco; pip install robosuite
- Run the following script: 
    - import robosuite
    from robosuite.environments.manipulation.stack import Stack
    env = Stack(robots="Panda") 
    obs = env.reset()

My traceback is:
`Exception ignored in: <function MjRenderContext.__del__ at 0x7fc780776790>
Traceback (most recent call last):
  File "/home/kxgao/study/kxgao.robosuite/robosuite/utils/binding_utils.py", line 199, in __del__
  File "/home/kxgao/study/kxgao.robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/kxgao/anaconda3/envs/robosuite/lib/python3.9/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
Exception ignored in: <function EGLGLContext.__del__ at 0x7fc7807765e0>
Traceback (most recent call last):
  File "/home/kxgao/study/kxgao.robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
  File "/home/kxgao/study/kxgao.robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/kxgao/anaconda3/envs/robosuite/lib/python3.9/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>`

Thank you for your help!

zhuyifengzju, 2022-12-21:
Hi,

   Is this an issue when you finish the script? I suspect this is because you are not calling the line `env.close()` at the end of the script.

Best,
Yifeng

kxgao, 2022-12-22:
Yes, adding `env.close()` indeed fixed it. Thank you!

closed 2022-12-22 by kxgao