ARISE-Initiative/robosuite #371: panda_arm_hand.urdf has the wrong origin for 'panda_joint1'

jstmn, 2022-12-17:
Hi,

I noticed that the panda_arm_hand.urdf urdf file has an incorrect value for the <origin> tag for 'panda_joint': https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/bullet_data/panda_description/urdf/panda_arm_hand.urdf#L43

Here's what the urdf looks like when visualized:
[image]

This can be fixed by changing `<origin rpy="0 0 0" xyz="0 -0.1 0"/>` to `<origin rpy="0 0 0" xyz="0 0 0.333"/>`

Happy to submit a pr for this 

Thanks,
Jeremy

zhuyifengzju, 2022-12-21:
Hi Jeremy,

     Thanks for finding this issue. Could you create a PR for this? Thanks!

Best,
Yifeng

jstmn, 2023-01-07:
Hi Yifeng,

Happy to do so. I have a local branch with the fixes, how do I submit them for a PR? I'm getting "ERROR: Permission to ARISE-Initiative/robosuite.git denied to jstmn." when I tried to push.

Thanks

jstmn, 2023-01-11:
Submitted as a pr here: https://github.com/ARISE-Initiative/robosuite/pull/375

closed 2023-04-13 by jstmn