ARISE-Initiative/robosuite #373: Using JOINT_POSITION Controller to send robot joints to desired absolute position?

ariana-spalter, 2022-12-21:
I am trying to use the JOINT_POSITION controller to move a robot's joints to a desired absolute joint position. I am using the Jaco in the Lift environment starting from the default arm position and trying to get the arm to have `joint_position_end = [4, 4, -0.59, 2, 1.1, 4, 0]` and have tried to do this in two different ways.

(1) Following #116, I was able to manually send position values using `env.robots[x].set_robot_joint_positions(joint_position_end)`, but this creates choppy transitions between joint positions which is limiting for eventually scaling up to hardware. I wanted to re-open #116 to ask if there is now a way for the joint position controller to interpret the input as absolute position commands?

(2) I also tried sending a delta position from the init_qpos of the Jaco to joint_position_end (`action_pos_delta = [0.808, 0.32, -0.59, 0.83, 1.05, 0.24, -3.142, 0]`) directly to the JOINT_POSITION controller, but this method also doesn’t result in getting to the desired absolute position exactly (`joint_position_end`). It was able to get close after sending the delta 100x through `env.step(action_pos_delta )` but there was no real control over it getting to the desired position as it was very easily able to overshoot `joint_position_end` if I let it run for just a few too many steps.

 Any help on controlling the joints to a desired position using the joint position controller would be greatly appreciated! Below please find the code I was using to test this.

````
from robosuite.controllers import load_controller_config
from robosuite.utils.input_utils import *

   if __name__ == "__main__":
    # Create dict to hold options that will be passed to env creation call
    options = {}

    # Choose environment and add it to options
    # options["env_name"] = choose_environment()
    options["env_name"] = 'Lift'
    controller_name = 'JOINT_POSITION'  # given desired positions for each joint change to torque
    # Load the desired controller
    options["controller_configs"] = load_controller_config(default_controller=controller_name)
    options["robots"] = 'Jaco'

    env = suite.make(
            **options,
            has_renderer=True,
            has_offscreen_renderer=False,
            ignore_done=True,
            use_camera_obs=False,
            control_freq=20,
            reward_shaping=True,
        )

    # Help message to user
    print()
    print("Press \"H\" to show the viewer control panel.")

    env.reset()
    env.viewer.set_camera(camera_id=0)

    # Get action limits
    low, high = env.action_spec

    for i in range(100):
        
        # (1):
        # joint_position_end = [4, 4, -0.59, 2, 1.1, 4, 0]
        # env.robots[0].set_robot_joint_positions(joint_position_end) #gripper not included
    
        # (2):
        # Jaco delta for JOINT_POSITION Controller:
        action_pos_delta = [0.808, 0.32, -0.59, 0.83, 1.05, 0.24, -3.142, -1]
        obs, reward, done, _ = env.step(action_pos_delta)

        env.render()`

zhuyifengzju, 2022-12-30:
Hi,
 In order to run the joint position controller, you would need to write your own control logic that takes care of the delta joint positions.  Here is a simple example for your reference:
    
```
def control_loop(tgt_jpos, env, max_n=50, eps=0.01):
    obs = env._get_observations()
    for i in range(max_n):
        q_diff_max = np.max(np.abs(obs["joint_pos"] - tgt_jpos))
        if q_diff_max < eps:
            break

        action = list(tgt_jpos) + list([-1])
        assert len(action) == 8, len(action)
        obs, _, _, _ = env.step(action)
        # env.render()

desired_joint_positions = # Your values go here
control_loop(desired_joint_positions, env)
```

Hope you will find this useful.

ariana-spalter, 2023-02-06:
Thank you for the suggestion!

I tried implementing this with some modifications but have been getting a high error of at least q_diff_max ~= 0.3 on average when I'm running the joint position controller, even if I make the max_n super high and sometimes even see a result where if I set the joints very close to the goal the joints will move away from the goal rather than towards. Is there anything you would suggest I do to try to fix this?

```
def control_loop(tgt_jpos, env, max_n=500, eps=0.01):
    obs = env._get_observations()
    for i in range(max_n):
        observed_value = obs[proprio_obs_name]
        if i == 0:
            observed_value = env.robots[0].init_qpos

        q_diff =  tgt_jpos - observed_value
        q_diff_max = np.max(np.abs(q_diff))
        if q_diff_max < eps:
            break

        action = list(q_diff) + list([1])
        assert len(action) == 8, len(action)
        obs, _, _, _ = env.step(action)

        #env.render()

 desired_joint_positions = # my desired joint positions
 obs = control_loop(desired_joint_positions, env) 
```

closed 2023-02-06 by ariana-spalter

zhuyifengzju, 2023-03-01:
I think it might require some tuning on the control gains of joint controllers: [https://github.com/ARISE-Initiative/robosuite/blob/60b358673cc5aaccbf1dc950dbfe365e0915c69a/robosuite/controllers/joint_pos.py#L240](url). Basically tuning the kp and kd gains to see if it can track the reference trajectory well.