ARISE-Initiative/robosuite #376: SubprocVectorEnv + Camera observation Error

GilgameshD, 2023-02-03:
Hi, I am trying to use vectorized environment with camera observation but I encountered the following error:

**mujoco.FatalError: Offscreen framebuffer is not complete, error 0x8cdd
Exception ignored in: <function MjRenderContext.__del__ at 0x7ff298536e50>**

If I don't use camera observation, everything is good. So is there anything else I need to do for rendering multiple mujoco environments？Thanks.

snasiriany, 2023-02-03:
This is a mujoco limitation rather than a robosuite-specific issue. I think you need to limit the number of mujoco environments that simultaneously render on the same gpu. I usually limit to 10 environments, that works fine for the gpus I use.

GilgameshD, 2023-02-04:
Thanks for your quick response! I finally find the cause of my problem. It seems not the number of the envs but some wired things. This is the test code I have：
```
import robosuite as robosuite
from robosuite.wrappers import GymWrapper
from tianshou.env import SubprocVectorEnv

env_param = {
    'robots': "Sawyer",  
    'use_camera_obs': True,
    'use_object_obs': False,
    'has_offscreen_renderer': True,
    'has_renderer': False, 
    'reward_shaping': True, 
    'control_freq': 20,  
}

dummy_env = GymWrapper(robosuite.make('Lift', **env_param))
train_envs = SubprocVectorEnv([lambda: GymWrapper(robosuite.make('Lift', **env_param)) for _ in range(1)])
```

It will cause the framebuffer error. However, if I comment out the dummy_env line, all things are good. Does that mean I should not use Env and VectorEnv together? Any explanation for this problem would be very helpful.

snasiriany, 2023-03-01:
Sorry for the late reply! For me I only use VectorEnv, but I still sometimes see this issue come up. Let us know if you see the issue again, closing for now.

closed 2023-03-01 by snasiriany