ARISE-Initiative/robosuite #381: MuJoCo Menagerie

kevinzakka, 2023-03-08:
Hey folks,

Wondering how you feel about transitioning the robot models to the ones in [Menagerie](https://github.com/deepmind/mujoco_menagerie). We'll be sysID'ing them later this year so they will keep improving.

snasiriany, 2023-03-13:
Hi Kevin, thanks for the suggestion! We haven't spoken as a team about this yet, here's my thoughts. If the Menagerie models are most realistic than the ones we have it'd be worth the transition. One potential challenge is maintaining backwards compatibility with our existing datasets (robomimic and other projects than use robosuite). So there's a tradeoff here. I'll let other team members add their thoughts as well.

kevinzakka, 2023-03-14:
That makes sense, thanks @snasiriany!

yukezhu, 2023-03-19:
Thanks @kevinzakka! Menagerie is a great effort and has been on our radar for a while. We decided not to migrate due to the backward compatibility issues Soroush mentioned earlier. But that's definite an option we should and will revisit for future robosuite releases.

closed 2023-03-19 by yukezhu