ARISE-Initiative/robosuite #382: Missing dependency in setup.py

wookayin, 2023-03-18:
It looks like h5py and termcolor a mandatory dependency, but it's not included in setup.py so installing and importing `robosuite` will result in an `ImportError` of `h5py` and `termcolor`.

snasiriany, 2023-04-07:
Thanks for letting us know. `termcolor` is [already included](https://github.com/ARISE-Initiative/robosuite/blob/1b825f11a937f5c18f2ac167af8ab084275fc625/setup.py#L25) in setup.py and `h5py` can be setup by running `pip install -r requirements-extra.txt`.

closed 2023-04-07 by snasiriany

wookayin, 2023-04-09:
Hi, but the package doesn't work out of box because h5py is missing. Robosuite is supposed to be installed via pip, and these packages are **mandatory** (otherwise you cannot import the module) -- requirements.txt is not a valid, standard way of listing dependencies.

snasiriany, 2023-04-12:
@wookayin thanks for following up. I'm don't think `h5py` in particular is mandatory to create environments and run them. Some scripts (mainly demos and data collection tools) require `h5py` but this is not something most users will need. Regarding requirement.txt being non-standard, I can run your concerns by my team, and if we decide to list everything in `setup.py` we will make the appropriate changes. Thanks!