ARISE-Initiative/robosuite #384: wrong comment at "models/objects/xml_objects.py"

ShuoZheLi, 2023-03-25:
`
def _set_door_damping(self, damping):
        """
        Helper function to override the door friction directly in the XML
        Args:
            damping (float): damping parameter to override the ones specified in the XML
        """
        hinge = find_elements(root=self.worldbody, tags="joint", attribs={"name": self.hinge_joint}, return_first=True)
        hinge.set("damping", array_to_string(np.array([damping])))
`
"door friction" should be "door damping"

yukezhu, 2023-03-25:
Hi @ShuoZheLi, do you mind submitting a PR to get this fixed? Please check our [contributing guidelines](https://github.com/ARISE-Initiative/robosuite/blob/master/CONTRIBUTING.md) for the process. Thanks!

ShuoZheLi, 2023-03-29:
Yes. There are numerous error like this in the code, I will collect them and create a PR. Thank you!