ARISE-Initiative/robosuite #385: SpaceMouse class seems to have bugs

tadashiK, 2023-03-25:
## Environments

- Ubuntu 22.04 with open-source Linux drivers for 3Dconnexion devices
- Using SpaceMouse Compact being detected by hid.enumerate().
- Robosuite installed in Conda's virtual environment
- Custom macro setup with correct device and vendor IDs

## Bug description

I am using SpaceMouse compact and encountered the following issue.

When I run `robosuite/robosuite/demos/demo_device_control.py`, hid fails to open my SpaceMouse Compact even if I set device and vendor IDs correctly in `macros_private.py`. This is weird because I was able to manually open it by `device = hid.device(); device.open(9583, 50741)`.

It turned out that SpaceMouse class uses default value of `macros.SPACEMOUSE_VENDOR_ID` and `macros.SPACEMOUSE_PRODUCT_ID` imported from `macros.py` instead of `macros_private.py`. If this is an intended behavior, it's fine. However, it is a bit weird to me that robosuite asks users to set up `macros_private.py` but does't use it internally.

Another lines that are seemingly a bug is the following;
```python
        while True:
            d = self.device.read(13)
            if d is not None and self._enabled:

                if macros.SPACEMOUSE_PRODUCT_ID == 50741:
                    ## logic for older spacemouse model
```
These lines check if `SPACEMOUSE_PRODUCT_ID` is 50741, but if SpaceMouse class is instantiated with `product_id=50741` without specifying `SPACEMOUSE_PRODUCT_ID` in `macros.py`, these line should check if `product_id` provided was 50741.

If these are indeed bugs, I will submit a PR to fix them.

tadashiK, 2023-04-01:
Also, another thing that seems to be a bug is that in both spacemouse and keyboard control cases, pressing ESC does not end the simulation. Indeed, the code does not seem to be detecting whether ESC key is pressed or not.

If you want me to fix this including above, please let me know if those are actually intended or indeed bugs. I can't do anything otherwise.

tadashiK, 2023-04-01:
I found [robosuite/devices/spacemouse.py](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/devices/spacemouse.py) also uses `import robosuite.macros as macros`, which loads the default `macros.py` instead of `private_macros.py`. Maybe the same issue exists in other files too?

snasiriany, 2023-04-07:
Thanks for raising all these issues! I've created a PR to address all of these here: https://github.com/ARISE-Initiative/robosuite/pull/389

snasiriany, 2023-04-07:
closing for now. let me know if you have more comments.

closed 2023-04-07 by snasiriany

tadashiK, 2023-05-14:
Hi, I am really sorry for the delayed response... The changes look great to me, thanks a lot! In particular, the way how you import private_maros is neat and better than what I did in an ad-hoc way :sweat_smile:

kunpeng-wu, 2024-03-21:
hello, may I ask you how to use spacemouse in Ubuntu, I cannot install the official driver, and hid cannot open the device

tadashiK, 2024-03-21:
Hi @kunpeng-wu , are you asking me? If you are, I answer your question although I haven't used robosuite for a while and almost forgot how I set up a spacemouse.

First of all, I followed instruction here: https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/devices/spacemouse.py

1. As far as I remember, I first installed a free version of a spacemouse driver (spacenavd) here: https://spacenav.sourceforge.net/
2. Then, I installed `hidapi` via pip.
3. Finally, I changed `SPACEMOUSE_VENDOR_ID` and `SPACEMOUSE_PRODUCT_ID` in `macros_private.py` to appropriate ones.

For finding appropriate vendor and product IDs, you may want to look into this website, where you can find vendor and product IDs of your spacemouse by searching with the name of your spacemouse: https://the-sz.com/products/usbid/

kavinwkp, 2024-03-22:
Thank you for your reply, I can use spacemouse through hidapi now. 😄

kunpeng-wu, 2024-03-22:
Thank you for your reply! I can use spacemouse through hidapi now. 😄