ARISE-Initiative/robosuite #388: Inconsistent behavior after calling MujocoEnv.reset when hard_reset is True

tadashiK, 2023-04-02:
It seems that when a MujocoEnv (and its child) is instantiated with `hard_reset=True`, its `_observables` attribute is inconsistent before and after calling its `reset` method.

Indeed, in the `__init__` method,

https://github.com/ARISE-Initiative/robosuite/blob/1b825f11a937f5c18f2ac167af8ab084275fc625/robosuite/environments/base.py#L145-L149

so, `self._observables` can be a returned value of `self.viewer._setup_observables()`. However, in the `reset` method,

https://github.com/ARISE-Initiative/robosuite/blob/1b825f11a937f5c18f2ac167af8ab084275fc625/robosuite/environments/base.py#L262-L264

so, `self._observables` is a returned value of `self._setup_observables()`.

Currently, this does not seem to be causing any issue, because all viewers implemented as of now (`OpenCVRenderer` and `NVISIIRenderer`) do not have a `_setup_observables` method. However, once a renderer that has a `_setup_observables` method is added, these lines would cause an issue, I think.

snasiriany, 2023-04-07:
thanks for bringing this up! Here's a PR implementing this: https://github.com/ARISE-Initiative/robosuite/pull/390
closing for now.

closed 2023-04-07 by snasiriany

tadashiK, 2023-05-14:
Thanks a lot! The change looks good to me too.