ARISE-Initiative/robosuite #397: Accessing rendered image during rollout

rayag, 2023-05-10:
Hi, I'm trying to use camera feedback as part of the environment observation on rollout. I would like to be able to render on the screen (requires on-screen renderer) and in the same time feed the camera feedback to a RL policy (requiring off-screen renderer), but both seem to be mutually exclusive. Is there an accepted way to do it?

Note: I noticed that I could use the `get_pixel_obs` function to get what the on-screen renderer has rendered. I'm using in my project the `MujocoPyRenderer` which does not implement this function. I have added a custom implementation: 
```python
def get_pixel_obs(self):
    resolution = glfw.get_framebuffer_size(self.viewer.sim._render_context_window.window)
    img = self.viewer.read_pixels(resolution[0], resolution[1], False)
    img = img[::-1, :, :] 
    return img
```
If you think that the implementation is correct and somebody may benefit from it in the future, then we could try to merge it.

_Robosuite is a great project, thank you!_

amandlek, 2023-05-12:
We should have support for simultaneous on-screen and off-screen rendering in robosuite v1.4, which uses the DeepMind MuJoCo bindings (not the OpenAI mujoco-py bindings). You can see the logic [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/environments/base.py#L288). If you're still using mujoco-py and an older version of robosuite, you can probably just adapt the OpenCVRenderer class (see [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/utils/opencv_renderer.py#L8)) so that you take offscreen rendered images and display them on screen as well. Hope this helps!

closed 2023-05-12 by amandlek

rayag, 2023-05-12:
Apparantly, I'm using version 1.3.3. I will upgrade to the latest version. Thank you for response!