ARISE-Initiative/robosuite #403: IK does not work in Panda

hai-h-nguyen, 2023-05-23:
I ran demo_device_control.py using a Panda and the Lift environment using IK control mode, but the performance was very bad. The robot oscillates a lot even after moving a bit using a keyboard (see the video below).

https://github.com/ARISE-Initiative/robosuite/assets/32955308/d57db150-9339-4b09-ba59-073319ef5ad7

hai-h-nguyen, 2023-05-23:
Turn out that the gain is too big https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/ik.py#L429.

closed 2023-05-23 by hai-h-nguyen

Steven-xzr, 2024-04-09:
Hi @hai-h-nguyen , I'm also attempting to use the IK controller. Thanks for your insightful finding. So what gain would you recommend if it's not -10.0 ?

Steven-xzr, 2024-04-09:
It seems that -0.5 works well.

Kimagure7, 2024-06-18:
> It seems that -0.5 works well.看来-0.5 效果很好。

thanks，this helps me.