ARISE-Initiative/robosuite #404: Changing colors of object during simulation

pvskand, 2023-05-23:
Thanks for maintaining this repo!

I was wondering if there's a way to change colors of objects during the simulation? Simply changing the `rgba` attribute of the `object` doesn't reflect during the rendering. I couldn't find anything about it in the documentation as well -- maybe I'm missing something.

snasiriany, 2023-06-26:
This guide from Mujoco may provide some insights: https://mujoco.readthedocs.io/en/latest/python.html?highlight=mjsim#named-access

You would need to follow the instructions in that guide, and likely change code here: https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/utils/binding_utils.py

Hope this is helpful!

pvskand, 2023-06-26:
Thanks @snasiriany. I'll check it out.

Closing the issue for now and will reopen if there are other concerns.

closed 2023-06-26 by pvskand