ARISE-Initiative/robosuite #405: Issue with creating Baxter environment

pratik2394, 2023-05-25:
I am following this page to create my own environment: https://robosuite.ai/docs/quickstart.html

I am trying to create an environment with Baxter robot with two ‘RethinkGripper’s..

Below is the code I am trying to run:

from robosuite.models import MujocoWorldBase

world = MujocoWorldBase()

from robosuite.models.robots import Baxter

mujoco_robot = Baxter()

from robosuite.models.grippers import gripper_factory

gripper = gripper_factory('RethinkGripper')
mujoco_robot.add_gripper(gripper)

mujoco_robot.set_base_xpos([0, 0, 0])
world.merge(mujoco_robot)

And it is throwing the following error:

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
[<ipython-input-20-26b5dbdff188>](https://macj32hlwl-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20230523-060150-RC00_534372501#) in <cell line: 12>()
     10 
     11 gripper = gripper_factory('RethinkGripper')
---> 12 mujoco_robot.add_gripper(gripper)
     13 
     14 mujoco_robot.set_base_xpos([0, 0, 0])

[/usr/local/lib/python3.10/dist-packages/robosuite/models/robots/manipulators/manipulator_model.py](https://macj32hlwl-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20230523-060150-RC00_534372501#) in add_gripper(self, gripper, arm_name)
     56         if arm_name is None:
     57             arm_name = self.eef_name
---> 58         if arm_name in self.grippers:
     59             raise ValueError("Attempts to add multiple grippers to one body")
     60 

TypeError: unhashable type: 'dict'

I tried multiple combinations… trying to provide dictionary with key words ‘right’, ‘left’, ‘right_hand’, ‘left_hand’ with values ‘RethinkGrippers’ but none of the combination seems to work. What am I missing?

awesome-aj0123, 2023-06-05:
Can you try `"robot0_right_hand"` or `"robot0_left_hand"`. I think for Baxter, these are the strings that you need to pass for arm_name. Let me know if this works.

Nuofan-SUSTech, 2024-05-07:
I tried `"robot0_right_hand"` or `"robot0_left_hand"`, but another error comes out:
```
File "/.conda/envs/robosuite/lib/python3.9/site-packages/robosuite/models/grippers/gripper_factory.py", line 26, in gripper_factory
    assert name in GRIPPER_MAPPING, f"Unknown gripper name: {name}. Valid options are: {ALL_GRIPPERS}"
AssertionError: Unknown gripper name: robot0_right_hand. Valid options are: dict_keys(['RethinkGripper', 'PandaGripper', 'JacoThreeFingerGripper', 'JacoThreeFingerDexterousGripper', 'WipingGripper', 'Robotiq85Gripper', 'Robotiq140Gripper', 'RobotiqThreeFingerGripper', 'RobotiqThreeFingerDexterousGripper', None])
```

Nuofan-SUSTech, 2024-05-07:
I tried to use:
```
mujoco_robot.add_gripper(gripper, "robot0_left_hand")
mujoco_robot.add_gripper(gripper, "robot0_right_hand")
```
solved the problem. I think `robot0_` is a prefix that add by robosuite, so it might depends on which robot id it is. And `left_hand`, `right_hand` is the name defined in the robot xml.

closed 2024-10-18 by kevin-thankyou-lin