ARISE-Initiative/robosuite #409: Install issue

warmmm, 2023-06-21:
Hi, when I use the cammand " python robosuite/demos/demo_random_action.py ", there is always happen the import bug, just like :

ImportError: Cannot initialize a EGL device display. This likely means that your EGL driver does not support the PLATFORM_DEVICE extension, which is required for creating a headless rendering context.

How can I slove this error?

dhanushTDS, 2023-07-25:
I am also facing the same issue, it would help a lot, if you provide support or clarity on how we can make it run devices with non-NVIDIA graphics cards (EGL not supported). Thanks

snasiriany, 2023-07-27:
@dhanushTDS can you add these lines to your `.bashrc` file, open a new terminal window, and try again?
```
export MUJOCO_GL=osmesa
export PYOPENGL_PLATFORM=osmesa
```

closed 2023-11-16 by zhuyifengzju