ARISE-Initiative/robosuite #415: Issue Running the Basic Demo code

dhanushTDS, 2023-07-27:
I happen to install robosuite both by PyPi and also from the source.

When i run the command :  python -m robosuite.demos.demo_random_action. It works as expected.

However, when i try doing from the git repo code, (robosuite) lm-2023@lm2023-XPS-8950:~/robosuite/robosuite/demos$ python demo_control.py

I face the following errror:
[image]

snasiriany, 2023-07-27:
Are you installing both the pypi version and source-based repos on the same virtual environment? That can lead to issues. The two should be installed in completely separate conda environments.

dhanushTDS, 2023-07-27:
This is what I had done, will try to uninstall the pypi version, and see if that fixes it

dhanushTDS, 2023-07-28:
On a completely other note, I have a small project which i need to implement quickly: 

Force-torque data guided RL for KUKA for an insertion problem. I already have the mesh/XML files for the insertion pieces, in the end I would like to simulate learning of a policy that can take in force-torque data and give position controls, in my case for the KUKA robot.

It would be great, if you could let me know, will this be possible on this platform, or I need to use some other library. Thanks

snasiriany, 2023-07-28:
I'm not sure. @zhuyifengzju @amandlek would you know?

amandlek, 2023-07-28:
I have not used force-torque data in robosuite unfortunately. Perhaps @roberto-martinmartin can chime in?

roberto-martinmartin, 2023-07-28:
Simulation of forces is in general quite difficult for any simulator. For Mujoco is even harder, since their contact model is optimized for soft impact, which is great for optimization purposes (e.g., more stable gradient descent) but is by far not realistic (contact forces start to act before the contact happens).
bottom line: Mujoco will give you some simulated forces that are ball-park ok (in the right order of magnitude) after you tune carefully the parameters (solimp, solref) of the involved bodies, but it won't realistic enough for 0-shot sim2real

roberto-martinmartin, 2023-07-28:
btw, I imagined the "force-torque data" you mentioned as input is a contact interface, e.g., what comes from a ft sensor at the wrist, not the joint torques

dhanushTDS, 2023-07-28:
Yes, i meant force torque sensor attached at the flange of the robot. I was wondering, can we make it so that, the force torque is gathered live from the actual robot and fed to the policy training on the robosuite simulation. 

You might ask why not just deploy without the use of a simulation software. It is just that, robosuite has the KUKA robot and its controller also. So i think its a convenient framework to test out RL algorithms and different reward functions.

Please feel free to give any kind of suggestions!

roberto-martinmartin, 2023-07-28:
sure, if you plug in a real ft signal and you train with it the forces will be real, but not connected to your sim model, though. but then training happens in the real world? that would be dangerous and slow. or do you want to train in sim but with real forces (still slow, bounded by real world)? in any case, using real world forces instead of the simulated ones decouples what is happening in sim and in real
good luck!

dhanushTDS, 2023-07-28:
Thanks for your insight, makes lot of sense!

closed 2023-07-28 by dhanushTDS