ARISE-Initiative/robosuite #416: Robosuite does not run on Windows with Python 3.11

ripkea, 2023-08-04:
Hi,

I'm using PyCharm with python 3.11 and want to run the simple robosuite example provided in the Quickstart. Whenever I try however, I get the error: 

"FileNotFoundError: Could not find module 'C:\Users\Me\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\robosuite\utils\mujoco.dll' (or one of its dependencies). Try using the full path with constructor syntax"

Which is true, there exists no such file. After some googling I think it may have to do with adding the path to find a .dll to PATH environment variable being deprecated after python 3.8 (found [here](https://stackoverflow.com/questions/59330863/cant-import-dll-module-in-python/64472088#64472088) I have of course tried to add mujoco to my PATH anyways and it did not work. And of course the latest version of mujoco is installed.

Is this a problem with the library and if yes, how can I remedy it? I don't necessarily want to switch to < 3.8 if it can be avoided.

Thank you so much!

yukezhu, 2023-08-04:
@awesome-aj0123 might have some experience running robosuite on Windows.

awesome-aj0123, 2023-08-07:
Which version of Windows are you currently using (i.e. 10 or 11)? I haven't installed Robosuite on Windows in a while and would like to go through the process again to see if I can recreate the same issues you ran into and see if I can fix it.

junaun, 2023-09-06:
I having similar issue when installing robosuite on anaconda python3.8 on Windows 10, this might be helpful: [https://github.com/ARISE-Initiative/robosuite/issues/323#issuecomment-1707864112](url)

closed 2023-12-29 by awesome-aj0123