ARISE-Initiative/robosuite #417: Saving and loading environment state at a specific point in time

MartinHVWinter, 2023-08-08:
Hello,

I'm currently working with robosuite v1.4 and have found it immensely useful for my research. However, I've encountered a challenge: I wish to save the state of an environment at a specific point in time and then be able to load back to that exact state at a later time.

Is there a built-in function or a recommended approach within robosuite to capture the current state of an environment and subsequently restore it at a later point? This would be greatly beneficial for me in order to test various algorithms under consistent conditions.

Thank you in advance for your assistance!

zhuyifengzju, 2023-08-08:
Hi,
I believe what you need is reading / loading the mujoco simulator states in the robosuite codebase. There is a way to do this in robosuite. To obtain the simulator state: `state=env.sim.get_state().flatten()`. And to load the simulator state: `env.sim.set_state_from_flattened(state)`

Note that there is some information that is non-markovian (for example, in the robot controller or some integrator terms in the mujoco simulator) that might not be saved in this state variable, so make sure that you are working with quasi-static situations in your use case.

MartinHVWinter, 2023-08-09:
Thank you for the quick response; it has helped me a lot.

closed 2023-08-09 by MartinHVWinter