ARISE-Initiative/robosuite #419: Add to documentation: How to create your own/custom environment and sensors?

sambaPython24, 2023-09-06:
The framework appears to be easy to use. But for many use cases, we would like to model our own environment
and our own sensors. 

Does anybody have experience with how to create your own environment/add your own sensors and could that be 
added to the [documentation](http://robosuite.ai/docs/v1.4/modules/environments.html) (with code)?

What was the pipeline to create the existing examples, do they base on known scene description file types such as ```.usd```, ```.urdf```,...?

zhuyifengzju, 2023-09-07:
Thanks for your praise in the framework! And thanks for the feedback! We are actually planning to update the documentation more to give better accessibility for advanced usage, like creating your own models. We will take your feedback into account when we upgrade the documentation. Please stay tuned! In the meantime, there are some available resources that have tried to make robosuite environments easier. For example, [our work](https://github.com/Lifelong-Robot-Learning/LIBERO) for procedural generation of tasks using PDDL language.

All mujoco related stuff are natively supporting `.mjcf` file format, which is also a XML style but different from `.urdf`. I think the mujoco team might have been developing the support for `.usd` and `.urdf`.

sambaPython24, 2023-09-22:
Ok, that would be interesting to know, because I also noticed that for at least two environments, the reward is almost always zero, such that the exploration phase takes very long until the task is randomly fulfilled at least once.

zhuyifengzju, 2023-09-26:
Which environments are you referring to?

Nimingez, 2023-12-04:
In wipe task, can I use my own table?