ARISE-Initiative/robosuite #422: How to deal with very sparse rewards

sambaPython24, 2023-09-22:
Hey, 

I noticed that for some of the more difficult environments ("TwoArmLift","TwoArmPegInHole","TwoArmHandover"), 
the reward is almost always zero (in fact i did not manage to get any reward). 

I can also not write my own reward function since I do not find the position of the objects such as a hammer or a box (everything except for the robotic degrees of freedom) in the returned value of either ```env.reset()```or ``env.step()```.

Maybe, one could get some reward after excessive exploration, but this is not very elegant. 

Do you have similar problems and how did you solve them?

junaun, 2023-10-01:
Did you set `reward_shaping=True` when creating environment? eg:
```
env = GymWrapper(suite.make(
    env_name="TwoArmPegInHole",
    robots=["UR5e","UR5e"], 
    has_renderer=True,
    has_offscreen_renderer=True,
    use_object_obs=False,                 
    use_camera_obs=True,
    camera_names="robot0_eye_in_hand",     
    camera_heights=84,                     
    camera_widths=84,                      
    reward_shaping=True,                    # use a dense reward signal for learning
    horizon = 500,
    control_freq=20,                      
))
```
You should be getting reward value around 200 at first epoch at horizon=500 for task "TwoArmPegInHole", similar to the [benchmark](https://robosuite.ai/docs/algorithms/benchmarking.html)
You can check out the reward shaping function in [two_arm_peg_in_hole.py](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/environments/manipulation/two_arm_peg_in_hole.py)

zhuyifengzju, 2023-11-16:
Feel free to reopen the issue if the reward issue continues.

closed 2023-11-16 by zhuyifengzju