ARISE-Initiative/robosuite #424: MjRenderContextOffscreen has no attribute 'con'

cynthia-you, 2023-10-10:
Hi, My v-env is python3.7 cuda11.1 mujoco210 robosuite1.4.0, but when i made a robosuite.make(), the error shows as: 'MjRenderContextOffscreen' object  has no attribute 'con'. Can you help me with that? thanx~

snasiriany, 2023-10-10:
please share with us a full stack trace so we can better help you. thanks!

cynthia-you, 2023-10-11:
[robosuite WARNING] No private macro file found! (__init__.py:7)
[robosuite WARNING] It is recommended to use a private macro file (__init__.py:8)
[robosuite WARNING] To setup, run: python /home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/scripts/setup_macros.py (__init__.py:9)
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
Traceback (most recent call last):
  File "/home/fusion/PycharmProjects/sawyer_analysis_reinforcement_learning-main/simple_srobosuite.py", line 18, in <module>
    use_camera_obs=False,
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/base.py", line 40, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/manipulation/lift.py", line 207, in __init__
    renderer_config=renderer_config,
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/manipulation/manipulation_env.py", line 186, in __init__
    renderer_config=renderer_config,
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/robot_env.py", line 226, in __init__
    renderer_config=renderer_config,
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/base.py", line 143, in __init__
    self._reset_internal()
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/manipulation/lift.py", line 389, in _reset_internal
    super()._reset_internal()
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/robot_env.py", line 510, in _reset_internal
    super()._reset_internal()
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/environments/base.py", line 296, in _reset_internal
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/utils/binding_utils.py", line 210, in __init__
    super().__init__(sim, offscreen=True, device_id=device_id, max_width=max_width, max_height=max_height)
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/utils/binding_utils.py", line 78, in __init__
    self.gl_ctx = GLContext(max_width=max_width, max_height=max_height, device_id=self.device_id)
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/renderers/context/egl_context.py", line 124, in __init__
    "Cannot initialize a EGL device display. This likely means that your EGL "
ImportError: Cannot initialize a EGL device display. This likely means that your EGL driver does not support the PLATFORM_DEVICE extension, which is required for creating a headless rendering context.
Exception ignored in: <function EGLGLContext.__del__ at 0x7f940c8070e0>
Traceback (most recent call last):
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/renderers/context/egl_context.py", line 155, in __del__
    self.free()
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/renderers/context/egl_context.py", line 146, in free
    if self._context:
AttributeError: 'EGLGLContext' object has no attribute '_context'
Exception ignored in: <function MjRenderContext.__del__ at 0x7f940c807290>
Traceback (most recent call last):
  File "/home/fusion/miniconda3/envs/robotrl/lib/python3.7/site-packages/robosuite/utils/binding_utils.py", line 198, in __del__
    self.con.free()
AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con'

Process finished with exit code 1

________________________________________________________________________________
Here's the full log

ma1ze, 2023-10-13:
Do you solve it? My env is python3.8 cuda12.1 mujoco210 robosuite1.4.0, meeting the same error.

zhuyifengzju, 2023-10-17:
Hi, can you provide more details on the hardware specs you are using? The log shows that you do not have EGL available for rendering, and that causes the offscreen rendering to crash.

zhuyifengzju, 2023-10-19:
Try `ldconfig -p | grep EGL`

cynthia-you, 2023-10-19:
> please share with us a full stack trace so we can better help you. thanks!
I have checked the EGL :
$ ldconfig -p | grep EGL
        libEGL_nvidia.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
        libEGL_nvidia.so.0 (libc6) => /usr/lib/i386-linux-gnu/libEGL_nvidia.so.0
        libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
        libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1
        libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so
______________________________________________________thanx____________________________________________________________________

zhuyifengzju, 2023-10-19:
It seems that you've installed EGL already. An issue from mujoco team's repo might be more relevant to your issue: https://github.com/google-deepmind/dm_control/issues/123

zhuyifengzju, 2023-11-16:
Closing this issue since it's inactive for a while. Feel free to reopen if the issue continues.

closed 2023-11-16 by zhuyifengzju

wangair, 2024-05-23:
Hi,I have the same problem.When I use the command "ldconfig -p | grep EGL", my system output is:
`libEGL_mesa.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_mesa.so.0
	libEGL.so.1 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so.1
	libEGL.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so
`
Does this mean that EGL is not installed correctly on my system?
Looking forward to your reply, thank you！ @zhuyifengzju

GuHuangAI, 2024-06-14:
@wangair Hello, I have the same problem. Have you solved it?

wangair, 2024-06-14:
> @wangair Hello, I have the same problem. Have you solved it?
Hi,I solved this just by reinstall nvidia driver.

balloch, 2024-06-28:
It is worth noting that I am able to make this error go away when it comes up by restarting the machine. This is not a long term solution, but it means that whatever is causing it-probably something to do with the EGL backend state-is sensitive to machine state.

pouyan-asg, 2025-12-04:
This helped me to run my code on a headless machine: MUJOCO_GL="glx" xvfb-run -s "-screen 0 1280x720x24" -a python XYZ.py 
or it can be updated with GPU as well: CUDA_VISIBLE_DEVICES=1 MUJOCO_GL="glx" xvfb-run -s "-screen 0 1280x720x24" -a python XYZ.py