ARISE-Initiative/robosuite #426: 'MjSim' object has no attribute 'model'

Soha18, 2023-10-12:
Hi,

I'm trying to use domain_randomization_wrapper with lift environment, but it through this error:
modder.save_defaults()
  File "~/lib/python3.10/site-packages/robosuite/utils/mjmod.py", line 867, in save_defaults
    self.textures = [Texture(self.model, i) for i in range(self.model.ntex)]
  File "~/lib/python3.10/site-packages/robosuite/utils/mjmod.py", line 60, in model
    return self.sim.model
AttributeError: 'MjSim' object has no attribute 'model'

would you please advice to solve it.

Thanks

zhuyifengzju, 2023-10-17:
Hi, can you give more context regarding your usage? Depending on the order of how you initialize the environment, the `MjSim` object might not have been created before querying the `model` attribute.

zhuyifengzju, 2023-10-19:
I see that you have your own implementation of `MultiModalEnv`. Have you tried not using it?

Soha18, 2023-10-19:
yes I mentioned that ,, even without applying my own wrapper the same error comes out..

babbatem, 2023-10-19:
I'm experiencing this error as well, with 'mujoco==2.3.7' and 'robosuite==1.4.0' from pip or source. The following script reproduces the error above:

`import numpy as np
import robosuite as suite
from robosuite.wrappers import DomainRandomizationWrapper

# create environment instance
env = suite.make(
    env_name="Door", # try with other tasks like "Stack" and "Door"
    robots="Panda",  # try with other robots like "Sawyer" and "Jaco"
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=False,
)
env = DomainRandomizationWrapper(env)

# reset the environment
env.reset()

for i in range(1000):
    action = np.random.randn(env.robots[0].dof) # sample random action
    obs, reward, done, info = env.step(action)  # take action in the environment
    env.render()  # render on display`

`Traceback (most recent call last):
  File "/Users/abba/projects/rma/test_free_space.py", line 16, in <module>
    env.reset()
  File "/Users/abba/miniconda3/envs/debugdomainrandom/lib/python3.11/site-packages/robosuite/wrappers/domain_randomization_wrapper.py", line 202, in reset
    self.save_default_domain()
  File "/Users/abba/miniconda3/envs/debugdomainrandom/lib/python3.11/site-packages/robosuite/wrappers/domain_randomization_wrapper.py", line 258, in save_default_domain
    modder.save_defaults()
  File "/Users/abba/miniconda3/envs/debugdomainrandom/lib/python3.11/site-packages/robosuite/utils/mjmod.py", line 866, in save_defaults
    self.textures = [Texture(self.model, i) for i in range(self.model.ntex)]
                                                           ^^^^^^^^^^
  File "/Users/abba/miniconda3/envs/debugdomainrandom/lib/python3.11/site-packages/robosuite/utils/mjmod.py", line 59, in model
    return self.sim.model
           ^^^^^^^^^^^^^^
AttributeError: 'MjSim' object has no attribute 'model'
`

babbatem, 2023-10-19:
Specifying `hard_reset=False` when creating the environment fixes the issue!

Soha18, 2023-10-19:
Hi,
Yes, initializing environment with hard_reset =False solved the problem.

Many thanks ,  you almost saved my life...

haru425, 2025-02-16:
Hi,

When I use DataCollectionWrapper, I faced the same problem.
Maybe [modder.update_sim](https://github.com/ARISE-Initiative/robosuite/blob/v1.5.0/robosuite/wrappers/domain_randomization_wrapper.py#L208) should come right after [super().reset()](https://github.com/ARISE-Initiative/robosuite/blob/v1.5.0/robosuite/wrappers/domain_randomization_wrapper.py#L199C15-L199C30)? Because MujocoEnv._destroy_sim will be called in super().reset().