ARISE-Initiative/robosuite #431: Error: eigenvalue of mesh inertia too small: robot0_link6_vis_7

aleada23, 2023-11-13:
Hy, I'm trying  to use demo_gym_functionality.py to test Robosuite. When I run the code using script with terminal I obtain the error message:

File "/home/user/lib/python3.10/site-packages/robosuite/utils/binding_utils.py", line 1072, in from_xml_string
    model = mujoco.MjModel.from_xml_string(xml)
ValueError: Error: eigenvalue of mesh inertia too small: robot0_link6_vis_7
Object name = robot0_link6_vis_7, id = 40, line = 121, column = -1.

With the previous demo I had no issues, but also with demo_device_control.py the same problem occurs.
I'm not able to solve the problem also looking for solutions on the web. Thank you

snasiriany, 2023-11-13:
if you swap this line to `robots="Panda",` will the issue go away? 

https://github.com/ARISE-Initiative/robosuite/blob/b9d8d3de5e3dfd1724f4a0e6555246c460407daa/robosuite/demos/demo_gym_functionality.py#L40

aleada23, 2023-11-13:
Sorry, the Error happens with Panda. If I use Sawyer as default I obtain ValueError: Error: mesh volume is too small: robot0_head_1. Which I think is almost the same issue

aleada23, 2023-11-14:
I tried to update Robosuite and Mujoco, it seems that the error no longer occurs. But, now a new problem arises trying to run the same file:
File "/home/user/demo_gym_functionality.py", line 56, in <module>
    observation, reward, terminated, truncated, info = env.step(action)
ValueError: not enough values to unpack (expected 5, got 4)

I tried to change versions of Mujoco or Gym, but I cannot fix this. Thank you.

zhuyifengzju, 2023-11-14:
You should do this instead:
```
observation, reward, terminated, debug_info = env.step(action)
```

zhuyifengzju, 2023-11-18:
We were not aware that Gym has changed its API to return five elements. step functions in Gym used to return four elements. Currently we will stick to the original design unless there are any necessary reasons to return five elements.

closed 2023-11-18 by zhuyifengzju