ARISE-Initiative/robosuite #434: Space Mouse macOS OSError: open failed

jsalfity, 2023-11-16:
I'm trying to use the Space Mouse on macOS. I should note I installed robosuite via pip, which doesn't seem to have the changes from this [PR](https://github.com/ARISE-Initiative/robosuite/pull/389), which sets the macros correctly and effects the `vendor_id` and `product_id`. I don't think my issue is related to that, I think my issue is perhaps some udev rule.
 
I followed instructions in `spacemouse.py`: 
https://github.com/ARISE-Initiative/robosuite/blob/b9d8d3de5e3dfd1724f4a0e6555246c460407daa/robosuite/devices/spacemouse.py#L6-L12

And I'm getting an error when trying to run `demo_device_control.py`, hitting
https://github.com/ARISE-Initiative/robosuite/blob/b9d8d3de5e3dfd1724f4a0e6555246c460407daa/robosuite/devices/spacemouse.py#L126

with error:
```sh
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "hid.pyx", line 142, in hid.device.open
OSError: open failed
```

Looks like my mac isn't able to read the device. However, when I use terminal to view devices with `hid.enumerate()`, I can see `com.3dconnexion` devices:
 ```py
{'path': b'DevSrvsID:4294970016', 'vendor_id': 9583, 'product_id': 50801, 'serial_number': 'com.3dconnexion.3DxMacCore-DriverKit-VirtualHIDMouse', 'release_number': 0, 'manufacturer_string': 'com.3dconnexion', 'product_string': '3Dconnexion Virtual Mouse', 'usage_page': 1, 'usage': 2, 'interface_number': -1}
{'path': b'DevSrvsID:4294970016', 'vendor_id': 9583, 'product_id': 50801, 'serial_number': 'com.3dconnexion.3DxMacCore-DriverKit-VirtualHIDMouse', 'release_number': 0, 'manufacturer_string': 'com.3dconnexion', 'product_string': '3Dconnexion Virtual Mouse', 'usage_page': 1, 'usage': 1, 'interface_number': -1}
{'path': b'DevSrvsID:4294970015', 'vendor_id': 1452, 'product_id': 592, 'serial_number': 'com.3dconnexion.3DxMacCore-DriverKit-VirtualHIDKeyboard', 'release_number': 0, 'manufacturer_string': 'com.3dconnexion', 'product_string': '3Dconnexion Virtual Keyboard', 'usage_page': 1, 'usage': 6, 'interface_number': -1}
{'path': b'DevSrvsID:4294970015', 'vendor_id': 1452, 'product_id': 592, 'serial_number': 'com.3dconnexion.3DxMacCore-DriverKit-VirtualHIDKeyboard', 'release_number': 0, 'manufacturer_string': 'com.3dconnexion', 'product_string': '3Dconnexion Virtual Keyboard', 'usage_page': 12, 'usage': 1, 'interface_number': -1}
{'path': b'DevSrvsID:4294970015', 'vendor_id': 1452, 'product_id': 592, 'serial_number': 'com.3dconnexion.3DxMacCore-DriverKit-VirtualHIDKeyboard', 'release_number': 0, 'manufacturer_string': 'com.3dconnexion', 'product_string': '3Dconnexion Virtual Keyboard', 'usage_page': 65280, 'usage': 1, 'interface_number': -1}
{'path': b'DevSrvsID:4294970015', 'vendor_id': 1452, 'product_id': 592, 'serial_number': 'com.3dconnexion.3DxMacCore-DriverKit-VirtualHIDKeyboard', 'release_number': 0, 'manufacturer_string': 'com.3dconnexion', 'product_string': '3Dconnexion Virtual Keyboard', 'usage_page': 65280, 'usage': 6, 'interface_number': -1}
```
so I know the device is being read.

I tried running the script as root with `sudo` but still the same problem. Is there any fancy udev rule I have to add?

zhuyifengzju, 2023-11-16:
Hi, this is what it looks like from my side when using `hid.enumerate`
```
 {'path': b'DevSrvsID:4295807982', 'vendor_id': 9583, 'product_id': 50734, 'serial_number': '', 'release_number': 1088, 'manufacturer_string': '3Dconnexion', 'product_string': 'SpaceMouse Wireless', 'usage_page': 1, 'usage': 8, 'interface_number': 0}, {'path': b'DevSrvsID:4295807982', 'vendor_id': 9583, 'product_id': 50734, 'serial_number': '', 'release_number': 1088, 'manufacturer_string': '3Dconnexion', 'product_string': 'SpaceMouse Wireless', 'usage_page': 1, 'usage': 48, 'interface_number': 0}
 ```
 Which version of 3D spacemouse do you have? We typically use the [wireless version](https://www.amazon.com/3Dconnexion-3DX-700059-Spacemouse-Compact-Mouse/dp/B079V4PXYD/ref=asc_df_B079V4PXYD/?tag=hyprod-20&linkCode=df0&hvadid=309773039951&hvpos=&hvnetw=g&hvrand=17810942560750023079&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9028319&hvtargid=pla-552724041547&psc=1&mcid=690e88c2f3a03046a1b57bb13212b8f6&gclid=Cj0KCQiAmNeqBhD4ARIsADsYfTca8pSKGIbS6s3vDyJmmCyxic-paa8kgNpahypHIrPL_YF_tR7gdCAaApWAEALw_wcB)

jsalfity, 2023-11-17:
Solved. The error is because I had the [3dconnexion](https://www.3dconnexion.com/service/drivers.html) driver installed.
When removing the driver from my MacBook I was able to open the device.

closed 2023-11-17 by jsalfity

oakwood-fujiken, 2025-04-24:
I have the same problem in Ubuntu 22.04. I use Spacemouse Compact (i.e. not wireless). 
I find the error when I run `python -m robocasa.demos.demo_teleop --device spacemouse`.

I have confirmed the following points: 
1. self.vender_id and self.product_id are collected. 
2. no other process is using the spacemouse 
3. change the `/etc/udev/rules.d/99-spacemouse.rules` to `SUBSYSTEM=="hidraw", ATTRS{idVendor}=="256f", ATTRS{idProduct}=="c635", MODE="0666"` and the permission of the spacemouse was collectively changed to 666.

If you have an idea to solve the problem or point out the cause of the problem, please let me know.

Any help would be greatly appreciated!