ARISE-Initiative/robosuite #437: OSC controller in control_delta =  False setting

sachinkundu, 2023-12-05:
Hello, this should have been basic but I cannot seem to make it work. I am trying to understand basic robosuite stuff including the OSC pose controller. I am instantiating Lift environment with Panda robot and OSC_POSE controller.

Now as I test to understand the coordinate systems used I just want to hold the position of the end effector. I use the following code.

```
import numpy as np
from robosuite.controllers import load_controller_config
from robosuite.utils.input_utils import *

if __name__ == "__main__":

    # Create dict to hold options that will be passed to env creation call
    options = {
        "robots": "Panda",
        "env_name": "Lift"
    }

    # Choose controller
    controller_name = "OSC_POSE"

    controller_config = load_controller_config(default_controller=controller_name)
    controller_config["control_delta"] = False

    # Load the desired controller
    options["controller_configs"] = controller_config

    # initialize the task
    env = suite.make(
        **options,
        has_renderer=True,
        has_offscreen_renderer=False,
        ignore_done=True,
        use_camera_obs=False,
        control_freq=20,
    )
    initial_state = env.reset()
    print(f"Initial eef_pose = {initial_state['robot0_eef_pos']}")
    env.viewer.set_camera(camera_id=0)

    # Get action limits
    low, high = env.action_spec
    action = np.append(initial_state['robot0_eef_pos'], [0, 0, 0, 1])
    # do visualization
    for i in range(1000):
        obs, reward, done, _ = env.step(action)
        env.render()
        # print(f"eef_pos: {obs['robot0_eef_pos']}")

```
I set the controller to work in absolute world coordinates as instructed on [OSC Controller Documentation](https://robosuite.ai/docs/simulation/controller.html#operation-space-controller) specifically `controller_config["control_delta"] = False`

After resetting the environment I get the cartesian pos of end effector frame and just feed the same post in the env step as action. I assumed that there will be no movement as I am working in absolute world coordinates and feed the same coordinate over and over.

However that does not happen and it seems to me that eef positon is moving not in a manner that I can predict.

Is there some demo pick and place sample where the robot is controlled. There is a random action demo but I would like to see the coordinates used(and transforms) for interacting with objects in the scene.

Entongsu, 2024-04-21:
I have encountered the same problems. Have you been able to figure it out? Thank you!

sachinkundu, 2024-05-19:
No. I just went with the delta option as the absolute world coordinates did not work for me.

omkarpatil18, 2024-08-10:
Hi, I think the robot should move as you specify `(0, 0, 0) `as the absolute axis angle, which is different from the initial? However, the `delta=False` setting did not work for me too. 

https://github.com/ARISE-Initiative/robosuite/issues/400#issuecomment-2276790526

kevin-thankyou-lin, 2024-10-30:
We've fixed this recently in v1.5.0 (now in the `master` branch): if you set `input_type=absolute` and `input_ref_frame=world`:

```
    from robosuite.controllers import load_composite_controller_config
    controller_config = load_composite_controller_config(robot="Panda")
    controller_config['body_parts']['right']['input_type'] = "absolute"
    controller_config['body_parts']['right']['input_ref_frame'] = "world"
```

and then set the action using:

```
action = np.zeros(7)
action[:3] = obs['robot0_eef_pos']
assert robosuite.__version__ == "1.5.0", "We use robot0_eef_quat_site for robosuite 1.5.0." \
                "Expected to be robot0_eef_quat for future versions."
action[3:6] = quat2axisangle(obs["robot0_eef_quat_site"])
```

the robot should stay there. 

Thanks for giving robosuite a try! Let us know if you've more questions.

closed 2024-11-06 by kevin-thankyou-lin