ARISE-Initiative/robosuite #443: Simultaneous Onscreen render and offscreen render

Dhanushvarma, 2023-12-19:
When trying to simultaneously call both render modes as follows:

` # HUMAN MODE PART #
            cam_id = self.env.sim.model.camera_name2id(camera_name)
            self.env.viewer.set_camera(cam_id)
            self.env.render()
            # RGB ARRAY PART #
            im = self.env.sim.render(height=height, width=width, camera_name=camera_name)
            import pdb; pdb.set_trace()
            return im[::-1]`

The render seems to be very choppy and the frames in the video appear to have be missing some frames. Can you point to how I can render on screen while simultaneously get the image from `env.sim`. Thanks

zhuyifengzju, 2023-12-29:
Hi @Dhanushvarma,
     Could you give a more concrete example? We want to understand if it's slow in rendering or slow in physics simulation.

Dhanushvarma, 2024-01-05:
What I have noticed is that, the frames that are being rendered (on-screen/ human mode), they seem to be jumping frames. 

My main goal is that, when I run the "run_trained_agent" script, I want it to render on-screen as well obtain the frame that is used for writing to the video file. Currently, the code seems to support, either of these at once. Hence, I tried to modify in such a way that it does both simultaneously:

current_frame = env.render(mode="test", height=1440, width=3440, camera_name=cam_name)

this method will now render on screen and also return the frame, but the aforementioned problem of skipping frames persists. I hope this is kind of clear.

zhuyifengzju, 2024-01-07:
first of all, if you turn on the image observation, you should be able to get the image sequence in `obs`. For example, if it's `agentview` camera, you can access them by `obs["agentview_image"]`

Second, I think your image rendering size is too large for efficient rendering. For policy training, we typically do `128x128` or `224x224`. I am not sure if what happens to you is actually "skipping  frames" --- to me it sounds like you are still experiencing slow rendering time so that the image sequence you see is discontinuous. If you still find it confusing, please send a video for us to review and see what is actually happening.

Dhanushvarma, 2024-01-13:
I am only using low-level observations. My main goal is to have a render method which returns the frame and as well renders it on screen, if that makes sense

zhuyifengzju, 2024-01-17:
Yes, the transformation seems right to me. You can render the actual point and double-check if the projection is correct.

As for the rendering, I still quite get the question --- you can enable camera observation and return the image array in `obs`, and use cv2 or its equivalence render it. If you are still experiencing the video rendering issue, please send in a video. Otherwise, it's very hard for us to analyze what is causing the issue.

Dhanushvarma, 2024-01-17:
As you mentioned that, I think it is the high resolution that is causing this issue, thanks for your help

closed 2024-01-17 by Dhanushvarma