ARISE-Initiative/robosuite #450: can't import shadowhand.xml into robosuite

Charlie0257, 2024-01-20:
Hi, @minhnh 

when I try to import [right_hand.xml](https://github.com/google-deepmind/mujoco_menagerie/blob/main/shadow_hand/right_hand.xml) as a gripper into robosuite, it always reports an error:

Traceback (most recent call last):
  File "/home/charlie/tempo/robosuite/robosuite/scripts/collect_human_demonstrations.py", line 284, in <module>
    env = suite.make(
  File "/home/charlie/tempo/robosuite/robosuite/environments/base.py", line 40, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/charlie/tempo/robosuite/robosuite/environments/manipulation/door.py", line 180, in __init__
    super().__init__(
  File "/home/charlie/tempo/robosuite/robosuite/environments/manipulation/manipulation_env.py", line 163, in __init__
    super().__init__(
  File "/home/charlie/tempo/robosuite/robosuite/environments/robot_env.py", line 219, in __init__
    super().__init__(
  File "/home/charlie/tempo/robosuite/robosuite/environments/base.py", line 134, in __init__
    self._load_model()
  File "/home/charlie/tempo/robosuite/robosuite/environments/manipulation/door.py", line 296, in _load_model
    super()._load_model()
  File "/home/charlie/tempo/robosuite/robosuite/environments/manipulation/single_arm_env.py", line 21, in _load_model
    super()._load_model()
  File "/home/charlie/tempo/robosuite/robosuite/environments/robot_env.py", line 353, in _load_model
    self._load_robots()
  File "/home/charlie/tempo/robosuite/robosuite/environments/robot_env.py", line 702, in _load_robots
    self.robots[idx].load_model()
  File "/home/charlie/tempo/robosuite/robosuite/robots/single_arm.py", line 155, in load_model
    self.gripper = gripper_factory(self.robot_model.default_gripper, idn=self.idn)
  File "/home/charlie/tempo/robosuite/robosuite/models/grippers/gripper_factory.py", line 29, in gripper_factory
    return GRIPPER_MAPPING[name](idn=idn)
  File "/home/charlie/tempo/robosuite/robosuite/models/grippers/shadow_hand.py", line 17, in __init__
    super().__init__(xml_path_completion("grippers/shadow_hand/right_hand.xml"), idn=idn)
  File "/home/charlie/tempo/robosuite/robosuite/models/grippers/gripper_model.py", line 21, in __init__
    super().__init__(fname, idn=idn)
  File "/home/charlie/tempo/robosuite/robosuite/models/base.py", line 496, in __init__
    super().__init__(fname)
  File "/home/charlie/tempo/robosuite/robosuite/models/base.py", line 47, in __init__
    self._replace_defaults_inline(default_dic=default_classes)
  File "/home/charlie/tempo/robosuite/robosuite/models/base.py", line 254, in _replace_defaults_inline
    self._replace_defaults_inline(default_dic=default_dic, root=child)
  File "/home/charlie/tempo/robosuite/robosuite/models/base.py", line 254, in _replace_defaults_inline
    self._replace_defaults_inline(default_dic=default_dic, root=child)
  File "/home/charlie/tempo/robosuite/robosuite/models/base.py", line 254, in _replace_defaults_inline
    self._replace_defaults_inline(default_dic=default_dic, root=child)
  File "/home/charlie/tempo/robosuite/robosuite/models/base.py", line 247, in _replace_defaults_inline
    tag_attrs = default_dic[cls_name].get(root.tag, None)
KeyError: 'wrist'

Process finished with exit code 1

-------------------------------------------------------------

Could you give me some suggestions?
Thanks for any help!

zhuyifengzju, 2024-01-24:
Hi, 
In order to use a new XML file, you need to carefully take care of the naming convention. For example, like [panda gripper ](https://github.com/ARISE-Initiative/robosuite/blob/b9d8d3de5e3dfd1724f4a0e6555246c460407daa/robosuite/models/assets/grippers/panda_gripper.xml#L14). You either needs to change the shadowhand.xml, or change the naming definition in python class in order to obtain correct result.

Jonas-Metzger, 2024-04-30:
where can I find the naming conventions?