ARISE-Initiative/robosuite #454: EGLerror while converting demonstrations with state obs to images

lakshitadodeja, 2024-02-20:
Hi, 

I am trying to get the demonstrations for the NutAssemblySquare environment using robomimic
In order to convert the low dimension observations in the demonstrations to images, we need to run the trajectories through robosuite. I am getting the following error while trying to do the same - 

```
Traceback (most recent call last):
  File "dataset_states_to_obs.py", line 425, in <module>
    dataset_states_to_obs(args)
  File "dataset_states_to_obs.py", line 260, in dataset_states_to_obs
    traj, camera_info = extract_trajectory(
  File "dataset_states_to_obs.py", line 129, in extract_trajectory
    r = env.get_reward()
  File "/home/dodeja/Documents/comps/robomimic/robomimic/envs/env_robosuite.py", line 335, in get_reward
    return self.env.reward()
  File "/home/dodeja/Documents/comps/robosuite/robosuite/environments/manipulation/nut_assembly.py", line 268, in reward
    self._check_success()
  File "/home/dodeja/Documents/comps/robosuite/robosuite/environments/manipulation/nut_assembly.py", line 640, in _check_success
    selff.objects_on_pegs[i] = int(self.on_peg(obj_pos, i) and r_reach < 0.6)
NameError: name 'selff' is not defined
Exception ignored in: <function MjRenderContext.__del__ at 0x7fec2717b940>
Traceback (most recent call last):
  File "/home/dodeja/Documents/comps/robosuite/robosuite/utils/binding_utils.py", line 199, in __del__
  File "/home/dodeja/Documents/comps/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/dodeja/miniconda3/envs/robosuite_2/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
Exception ignored in: <function EGLGLContext.__del__ at 0x7fec2717b790>
Traceback (most recent call last):
  File "/home/dodeja/Documents/comps/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
  File "/home/dodeja/Documents/comps/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/dodeja/miniconda3/envs/robosuite_2/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>

```Traceback (most recent call last):
  File "dataset_states_to_obs.py", line 425, in <module>
    dataset_states_to_obs(args)
  File "dataset_states_to_obs.py", line 260, in dataset_states_to_obs
    traj, camera_info = extract_trajectory(
  File "dataset_states_to_obs.py", line 129, in extract_trajectory
    r = env.get_reward()
  File "/home/dodeja/Documents/comps/robomimic/robomimic/envs/env_robosuite.py", line 335, in get_reward
    return self.env.reward()
  File "/home/dodeja/Documents/comps/robosuite/robosuite/environments/manipulation/nut_assembly.py", line 268, in reward
    self._check_success()
  File "/home/dodeja/Documents/comps/robosuite/robosuite/environments/manipulation/nut_assembly.py", line 640, in _check_success
    selff.objects_on_pegs[i] = int(self.on_peg(obj_pos, i) and r_reach < 0.6)
NameError: name 'selff' is not defined
Exception ignored in: <function MjRenderContext.__del__ at 0x7fec2717b940>
Traceback (most recent call last):
  File "/home/dodeja/Documents/comps/robosuite/robosuite/utils/binding_utils.py", line 199, in __del__
  File "/home/dodeja/Documents/comps/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/dodeja/miniconda3/envs/robosuite_2/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
Exception ignored in: <function EGLGLContext.__del__ at 0x7fec2717b790>
Traceback (most recent call last):
  File "/home/dodeja/Documents/comps/robosuite/robosuite/renderers/context/egl_context.py", line 155, in __del__
  File "/home/dodeja/Documents/comps/robosuite/robosuite/renderers/context/egl_context.py", line 149, in free
  File "/home/dodeja/miniconda3/envs/robosuite_2/lib/python3.8/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
```

I added hard_reset=false while creating the environment and I am able to see normal rendering using demo_random_action.py

lakshitadodeja, 2024-05-15:
I just downloaded the image data from here -
robomimic -  https://robomimic.github.io/docs/v0.2/datasets/robomimic_v0.1.html 
mimicgen -  https://drive.google.com/drive/folders/14e9kkHGfApuQ709LBEbXrXVI1Lp5Ax7p

pengzhi1998, 2024-05-16:
Got it. Thank you for your reply and help!!

closed 2024-09-07 by kevin-thankyou-lin