ARISE-Initiative/robosuite #457: How can I find the dh parameters for the two arms of baxter robot?

aleada23, 2024-03-03:
Good morning.
I'm trying to derive the dh parameters ,in order to apply control using the Jacobian, of the two arms with baxter robot. However, if I use the dh method to compute the direct kinematic of the arms, in order to check the correctness of my calculations, I obtain a different position from the one provided using the property  _hand_pos. I tried different parameters for my dh table founded online, and then I tried to derive the parametrs from the xml file of the robot. In both cases the position is different from the real one. I would like to ask if someone can provide me the parameters in order to obtain the dh table or give me a method that works to find them.

Thank you

zhuyifengzju, 2024-03-06:
Hi, would you provide more information regarding this? It seems like what you really want is the jacobian values. In that case, I don't think you actually need a dh table, and can get the values by using the mujoco basic functions.

aleada23, 2024-03-06:
Yes, I'm interested in Jacobian values, but I need the transformation matrices from the base of robot for all the links too. Is there a way to obtain this information with joint values as variables?
Thanks

zhuyifengzju, 2024-03-07:
Do you think this [tutorial notebook](https://github.com/ARISE-Initiative/robosuite-notebooks/blob/master/kinematics.ipynb) contains all the information you would need?

aleada23, 2024-03-07:
Yes, I have everything that I need for the moment. 
Thank you so much, it's my first time with Mujoco and Robosuite and this notebook is really useful

closed 2024-03-07 by zhuyifengzju