ARISE-Initiative/robosuite #458: How to create your own device?

Marshall-Jericho, 2024-03-06:


zhuyifengzju, 2024-03-06:
Hi, can you provide more information about this? What kind of device do you want to use? As long as you can get an array of action values, integrating your custom device should be fairly easy.

Marshall-Jericho, 2024-03-07:
My device is omega.7, which can feed back six-axis data in real time. When using this module to run demo_device_control.py, the input2action function seems to have some problems, and the frequency of image switching per frame is very low.
below is my code
`import threading
import time 

import numpy as np

import omega

from robosuite.devices import Device
from robosuite.utils.transform_utils import rotation_matrix

class Omega7(Device):
    def __init__(
        self,
        pos_sensitivity=400.0,
        rot_sensitivity=1.0,
    ):
        print("Opening Omega7 device")
        omega.open_device()
        self.pos_sensitivity = pos_sensitivity
        self.rot_sensitivity = rot_sensitivity
        # 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.last_x, self.last_y, self.last_z = 0, 0, 0

        self.roll, self.pitch, self.yaw = 0, 0, 0
        self.last_roll, self.last_pitch, self.last_yaw = 0, 0, 0
        self.single_click_and_hold = False

        self._control = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._lastcontrol = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._reset_state = 0
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        self._enabled = False

        # launch a new listener thread to listen to SpaceMouse
        self.thread = threading.Thread(target=self.run)
        self.thread.daemon = True
        self.thread.start()

    def _reset_internal_state(self):
        """
        Resets internal state of controller, except for the reset signal.
        """
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        # Reset 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.roll, self.pitch, self.yaw = 0, 0, 0
        # Reset control
        self._control = np.zeros(6)
        # Reset grasp
        self.single_click_and_hold = False

    def start_control(self):
        """
        Method that should be called externally before controller can
        start receiving commands.
        """
        self._reset_internal_state()
        self._reset_state = 0
        self._enabled = True

    def get_controller_state(self):
        """
        Grabs the current state of the 3D mouse.

        Returns:
            dict: A dictionary containing dpos, orn, unmodified orn, grasp, and reset
        """
        dpos = (self.control - self._lastcontrol)[:3] * 0.005 * self.pos_sensitivity
        # dpos = self.control[:3] * 0.005 * self.pos_sensitivity # 此处0.005可能要改        
        # roll, pitch, yaw = self.control[3:] * 0.005 * self.rot_sensitivity
        roll, pitch, yaw = 0 , 0,  0
        self.last_x, self.last_y, self.last_z,self.last_roll, self.last_pitch, self.last_yaw = self.x,self.y,self.z,self.roll,self.pitch,self.yaw
        self._lastcontrol = [
                self.last_x,
                self.last_y,
                self.last_z,
                self.last_roll,
                self.last_pitch,
                self.last_yaw,
            ]
        # convert RPY to an absolute orientation
        drot1 = rotation_matrix(angle=-pitch, direction=[1.0, 0, 0], point=None)[:3, :3]
        drot2 = rotation_matrix(angle=roll, direction=[0, 1.0, 0], point=None)[:3, :3]
        drot3 = rotation_matrix(angle=yaw, direction=[0, 0, 1.0], point=None)[:3, :3]

        self.rotation = self.rotation.dot(drot1.dot(drot2.dot(drot3)))

        return dict(
            dpos=dpos,
            rotation=self.rotation,
            raw_drotation=np.array([roll, pitch, yaw]),
            grasp=self.control_gripper,
            reset=self._reset_state,
        )
    

    def run(self):
        while True:
            self.x,self.y, self.z,self.roll,self.pitch,self.yaw = omega.get_pos_and_orideg()
            self._control = [
                self.x,
                self.y,
                self.z,
                self.roll,
                self.pitch,
                self.yaw,
            ]
     

        # 夹爪功能待完善

    @property
    def control(self):
        """
        Grabs current pose of Spacemouse

        Returns:
            np.array: 6-DoF control value
        """
        return np.array(self._control)

    @property
    def control_gripper(self):
        """
        Maps internal states into gripper commands.

        Returns:
            float: Whether we're using single click and hold or not
        """
        if self.single_click_and_hold:
            return 1.0
        return 0
    

# if __name__ == "__main__":

#     omega7 = Omega7()
#     for i in range(100):
#     # while True:
#         print(omega7.control, omega7.control_gripper)
#         time.sleep(0.02)
if __name__ == "__main__":
    omega7 = Omega7()
    for i in range(100):  # 或者使用 while True: 以持续运行
        # 获取控制器状态
        controller_state = omega7.get_controller_state()
        # 仅打印 dpos 值
        print("dpos:", controller_state["dpos"])
        time.sleep(0.02)  # 控制循环速度，以便更容易阅读输出
`

Marshall-Jericho, 2024-03-07:
My device is omega.7, which can feed back six-axis data in real time. When using this module to run demo_device_control.py, the input2action function seems to have some problems, and the frequency of image switching per frame is very low.
my code is follow
`
import threading
import time 

import numpy as np

import omega

from robosuite.devices import Device
from robosuite.utils.transform_utils import rotation_matrix

class Omega7(Device):
    def __init__(
        self,
        pos_sensitivity=400.0,
        rot_sensitivity=1.0,
    ):
        print("Opening Omega7 device")
        omega.open_device()
        self.pos_sensitivity = pos_sensitivity
        self.rot_sensitivity = rot_sensitivity
        # 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.last_x, self.last_y, self.last_z = 0, 0, 0

        self.roll, self.pitch, self.yaw = 0, 0, 0
        self.last_roll, self.last_pitch, self.last_yaw = 0, 0, 0
        self.single_click_and_hold = False

        self._control = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._lastcontrol = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._reset_state = 0
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        self._enabled = False

        # launch a new listener thread to listen to SpaceMouse
        self.thread = threading.Thread(target=self.run)
        self.thread.daemon = True
        self.thread.start()

    def _reset_internal_state(self):
        """
        Resets internal state of controller, except for the reset signal.
        """
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        # Reset 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.roll, self.pitch, self.yaw = 0, 0, 0
        # Reset control
        self._control = np.zeros(6)
        # Reset grasp
        self.single_click_and_hold = False

    def start_control(self):
        """
        Method that should be called externally before controller can
        start receiving commands.
        """
        self._reset_internal_state()
        self._reset_state = 0
        self._enabled = True

    def get_controller_state(self):
        """
        Grabs the current state of the 3D mouse.

        Returns:
            dict: A dictionary containing dpos, orn, unmodified orn, grasp, and reset
        """
        dpos = (self.control - self._lastcontrol)[:3] * 0.005 * self.pos_sensitivity
        # dpos = self.control[:3] * 0.005 * self.pos_sensitivity # 此处0.005可能要改        
        # roll, pitch, yaw = self.control[3:] * 0.005 * self.rot_sensitivity
        roll, pitch, yaw = 0 , 0,  0
        self.last_x, self.last_y, self.last_z,self.last_roll, self.last_pitch, self.last_yaw = self.x,self.y,self.z,self.roll,self.pitch,self.yaw
        self._lastcontrol = [
                self.last_x,
                self.last_y,
                self.last_z,
                self.last_roll,
                self.last_pitch,
                self.last_yaw,
            ]
        # convert RPY to an absolute orientation
        drot1 = rotation_matrix(angle=-pitch, direction=[1.0, 0, 0], point=None)[:3, :3]
        drot2 = rotation_matrix(angle=roll, direction=[0, 1.0, 0], point=None)[:3, :3]
        drot3 = rotation_matrix(angle=yaw, direction=[0, 0, 1.0], point=None)[:3, :3]

        self.rotation = self.rotation.dot(drot1.dot(drot2.dot(drot3)))

        return dict(
            dpos=dpos,
            rotation=self.rotation,
            raw_drotation=np.array([roll, pitch, yaw]),
            grasp=self.control_gripper,
            reset=self._reset_state,
        )
    

    def run(self):
        while True:
            self.x,self.y, self.z,self.roll,self.pitch,self.yaw = omega.get_pos_and_orideg()
            self._control = [
                self.x,
                self.y,
                self.z,
                self.roll,
                self.pitch,
                self.yaw,
            ]
     

        # 夹爪功能待完善

    @property
    def control(self):
        """
        Grabs current pose of Spacemouse

        Returns:
            np.array: 6-DoF control value
        """
        return np.array(self._control)

    @property
    def control_gripper(self):
        """
        Maps internal states into gripper commands.

        Returns:
            float: Whether we're using single click and hold or not
        """
        if self.single_click_and_hold:
            return 1.0
        return 0
    

# if __name__ == "__main__":

#     omega7 = Omega7()
#     for i in range(100):
#     # while True:
#         print(omega7.control, omega7.control_gripper)
#         time.sleep(0.02)
if __name__ == "__main__":
    omega7 = Omega7()
    for i in range(100):  # 或者使用 while True: 以持续运行
        # 获取控制器状态
        controller_state = omega7.get_controller_state()
        # 仅打印 dpos 值
        print("dpos:", controller_state["dpos"])
        time.sleep(0.02)  # 控制循环速度，以便更容易阅读输出

`

Marshall-Jericho, 2024-03-07:
My device is omega.7, which can feed back six-axis data in real time. When using this module to run demo_device_control.py, the input2action function seems to have some problems, and the frequency of image switching per frame is very low.
my code is follow
import threading
import time 

import numpy as np

import omega

from robosuite.devices import Device
from robosuite.utils.transform_utils import rotation_matrix

class Omega7(Device):
    def __init__(
        self,
        pos_sensitivity=400.0,
        rot_sensitivity=1.0,
    ):
        print("Opening Omega7 device")
        omega.open_device()
        self.pos_sensitivity = pos_sensitivity
        self.rot_sensitivity = rot_sensitivity
        # 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.last_x, self.last_y, self.last_z = 0, 0, 0

        self.roll, self.pitch, self.yaw = 0, 0, 0
        self.last_roll, self.last_pitch, self.last_yaw = 0, 0, 0
        self.single_click_and_hold = False

        self._control = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._lastcontrol = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._reset_state = 0
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        self._enabled = False

        # launch a new listener thread to listen to SpaceMouse
        self.thread = threading.Thread(target=self.run)
        self.thread.daemon = True
        self.thread.start()

    def _reset_internal_state(self):
        """
        Resets internal state of controller, except for the reset signal.
        """
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        # Reset 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.roll, self.pitch, self.yaw = 0, 0, 0
        # Reset control
        self._control = np.zeros(6)
        # Reset grasp
        self.single_click_and_hold = False

    def start_control(self):
        """
        Method that should be called externally before controller can
        start receiving commands.
        """
        self._reset_internal_state()
        self._reset_state = 0
        self._enabled = True

    def get_controller_state(self):
        """
        Grabs the current state of the 3D mouse.

        Returns:
            dict: A dictionary containing dpos, orn, unmodified orn, grasp, and reset
        """
        dpos = (self.control - self._lastcontrol)[:3] * 0.005 * self.pos_sensitivity
        # dpos = self.control[:3] * 0.005 * self.pos_sensitivity # 此处0.005可能要改        
        # roll, pitch, yaw = self.control[3:] * 0.005 * self.rot_sensitivity
        roll, pitch, yaw = 0 , 0,  0
        self.last_x, self.last_y, self.last_z,self.last_roll, self.last_pitch, self.last_yaw = self.x,self.y,self.z,self.roll,self.pitch,self.yaw
        self._lastcontrol = [
                self.last_x,
                self.last_y,
                self.last_z,
                self.last_roll,
                self.last_pitch,
                self.last_yaw,
            ]
        # convert RPY to an absolute orientation
        drot1 = rotation_matrix(angle=-pitch, direction=[1.0, 0, 0], point=None)[:3, :3]
        drot2 = rotation_matrix(angle=roll, direction=[0, 1.0, 0], point=None)[:3, :3]
        drot3 = rotation_matrix(angle=yaw, direction=[0, 0, 1.0], point=None)[:3, :3]

        self.rotation = self.rotation.dot(drot1.dot(drot2.dot(drot3)))

        return dict(
            dpos=dpos,
            rotation=self.rotation,
            raw_drotation=np.array([roll, pitch, yaw]),
            grasp=self.control_gripper,
            reset=self._reset_state,
        )
    

    def run(self):
        while True:
            self.x,self.y, self.z,self.roll,self.pitch,self.yaw = omega.get_pos_and_orideg()
            self._control = [
                self.x,
                self.y,
                self.z,
                self.roll,
                self.pitch,
                self.yaw,
            ]
     

        # 夹爪功能待完善

    @property
    def control(self):
        """
        Grabs current pose of Spacemouse

        Returns:
            np.array: 6-DoF control value
        """
        return np.array(self._control)

    @property
    def control_gripper(self):
        """
        Maps internal states into gripper commands.

        Returns:
            float: Whether we're using single click and hold or not
        """
        if self.single_click_and_hold:
            return 1.0
        return 0
    

# if __name__ == "__main__":

#     omega7 = Omega7()
#     for i in range(100):
#     # while True:
#         print(omega7.control, omega7.control_gripper)
#         time.sleep(0.02)
if __name__ == "__main__":
    omega7 = Omega7()
    for i in range(100):  # 或者使用 while True: 以持续运行
        # 获取控制器状态
        controller_state = omega7.get_controller_state()
        # 仅打印 dpos 值
        print("dpos:", controller_state["dpos"])
        time.sleep(0.02)  # 控制循环速度，以便更容易阅读输出
``

Marshall-Jericho, 2024-03-07:
‘’‘
dfsfsd
'''

Marshall-Jericho, 2024-03-07:
Some basic Git commands are:
```
git status
git add
git commit
```

Marshall-Jericho, 2024-03-07:
My device is omega.7, which can feed back six-axis data in real time (absolute quantity rather than variation), but when I use this module I wrote in demo_device_control.py, there may be some problems when executing input2action() (I don’t OK), the phenomenon is: 1. The simulation is stuck 2. The generated action is unreasonable。
Thank you very much for your help
```
import threading
import time 

import numpy as np

import omega

from robosuite.devices import Device
from robosuite.utils.transform_utils import rotation_matrix

class Omega7(Device):
    def __init__(
        self,
        pos_sensitivity=400.0,
        rot_sensitivity=1.0,
    ):
        print("Opening Omega7 device")
        omega.open_device()
        self.pos_sensitivity = pos_sensitivity
        self.rot_sensitivity = rot_sensitivity
        # 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.last_x, self.last_y, self.last_z = 0, 0, 0

        self.roll, self.pitch, self.yaw = 0, 0, 0
        self.last_roll, self.last_pitch, self.last_yaw = 0, 0, 0
        self.single_click_and_hold = False

        self._control = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._lastcontrol = np.array([0.0, 0.0, 0.0, 0.0, 0.0, 0.0])
        self._reset_state = 0
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        self._enabled = False

        # launch a new listener thread to listen to SpaceMouse
        self.thread = threading.Thread(target=self.run)
        self.thread.daemon = True
        self.thread.start()

    def _reset_internal_state(self):
        """
        Resets internal state of controller, except for the reset signal.
        """
        self.rotation = np.array([[-1.0, 0.0, 0.0], [0.0, 1.0, 0.0], [0.0, 0.0, -1.0]])
        # Reset 6-DOF variables
        self.x, self.y, self.z = 0, 0, 0
        self.roll, self.pitch, self.yaw = 0, 0, 0
        # Reset control
        self._control = np.zeros(6)
        # Reset grasp
        self.single_click_and_hold = False

    def start_control(self):
        """
        Method that should be called externally before controller can
        start receiving commands.
        """
        self._reset_internal_state()
        self._reset_state = 0
        self._enabled = True

    def get_controller_state(self):
        """
        Grabs the current state of the 3D mouse.

        Returns:
            dict: A dictionary containing dpos, orn, unmodified orn, grasp, and reset
        """
        dpos = (self.control - self._lastcontrol)[:3] * 0.005 * self.pos_sensitivity
        # dpos = self.control[:3] * 0.005 * self.pos_sensitivity # 此处0.005可能要改        
        # roll, pitch, yaw = self.control[3:] * 0.005 * self.rot_sensitivity
        roll, pitch, yaw = 0 , 0,  0
        self.last_x, self.last_y, self.last_z,self.last_roll, self.last_pitch, self.last_yaw = self.x,self.y,self.z,self.roll,self.pitch,self.yaw
        self._lastcontrol = [
                self.last_x,
                self.last_y,
                self.last_z,
                self.last_roll,
                self.last_pitch,
                self.last_yaw,
            ]
        # convert RPY to an absolute orientation
        drot1 = rotation_matrix(angle=-pitch, direction=[1.0, 0, 0], point=None)[:3, :3]
        drot2 = rotation_matrix(angle=roll, direction=[0, 1.0, 0], point=None)[:3, :3]
        drot3 = rotation_matrix(angle=yaw, direction=[0, 0, 1.0], point=None)[:3, :3]

        self.rotation = self.rotation.dot(drot1.dot(drot2.dot(drot3)))

        return dict(
            dpos=dpos,
            rotation=self.rotation,
            raw_drotation=np.array([roll, pitch, yaw]),
            grasp=self.control_gripper,
            reset=self._reset_state,
        )
    

    def run(self):
        while True:
            self.x,self.y, self.z,self.roll,self.pitch,self.yaw = omega.get_pos_and_orideg()
            self._control = [
                self.x,
                self.y,
                self.z,
                self.roll,
                self.pitch,
                self.yaw,
            ]
     

        # 夹爪功能待完善

    @property
    def control(self):
        """
        Grabs current pose of Spacemouse

        Returns:
            np.array: 6-DoF control value
        """
        return np.array(self._control)

    @property
    def control_gripper(self):
        """
        Maps internal states into gripper commands.

        Returns:
            float: Whether we're using single click and hold or not
        """
        if self.single_click_and_hold:
            return 1.0
        return 0
    

# if __name__ == "__main__":

#     omega7 = Omega7()
#     for i in range(100):
#     # while True:
#         print(omega7.control, omega7.control_gripper)
#         time.sleep(0.02)
if __name__ == "__main__":
    omega7 = Omega7()
    for i in range(100):  # 或者使用 while True: 以持续运行
        # 获取控制器状态
        controller_state = omega7.get_controller_state()
        # 仅打印 dpos 值
        print("dpos:", controller_state["dpos"])
        time.sleep(0.02)  # 控制循环速度，以便更容易阅读输出

```

Marshall-Jericho, 2024-03-07:
```
"""Teleoperate robot with keyboard or SpaceMouse.

***Choose user input option with the --device argument***

Keyboard:
    We use the keyboard to control the end-effector of the robot.
    The keyboard provides 6-DoF control commands through various keys.
    The commands are mapped to joint velocities through an inverse kinematics
    solver from Bullet physics.

    Note:
        To run this script with macOS, you must run it with root access.

SpaceMouse:

    We use the SpaceMouse 3D mouse to control the end-effector of the robot.
    The mouse provides 6-DoF control commands. The commands are mapped to joint
    velocities through an inverse kinematics solver from Bullet physics.

    The two side buttons of SpaceMouse are used for controlling the grippers.

    SpaceMouse Wireless from 3Dconnexion: https://www.3dconnexion.com/spacemouse_wireless/en/
    We used the SpaceMouse Wireless in our experiments. The paper below used the same device
    to collect human demonstrations for imitation learning.

    Reinforcement and Imitation Learning for Diverse Visuomotor Skills
    Yuke Zhu, Ziyu Wang, Josh Merel, Andrei Rusu, Tom Erez, Serkan Cabi, Saran Tunyasuvunakool,
    János Kramár, Raia Hadsell, Nando de Freitas, Nicolas Heess
    RSS 2018

    Note:
        This current implementation only supports macOS (Linux support can be added).
        Download and install the driver before running the script:
            https://www.3dconnexion.com/service/drivers.html

Additionally, --pos_sensitivity and --rot_sensitivity provide relative gains for increasing / decreasing the user input
device sensitivity

***Choose controller with the --controller argument***

Choice of using either inverse kinematics controller (ik) or operational space controller (osc):
Main difference is that user inputs with ik's rotations are always taken relative to eef coordinate frame, whereas
    user inputs with osc's rotations are taken relative to global frame (i.e.: static / camera frame of reference).

    Notes:
        OSC also tends to be more computationally efficient since IK relies on the backend pybullet IK solver.

***Choose environment specifics with the following arguments***

    --environment: Task to perform, e.g.: "Lift", "TwoArmPegInHole", "NutAssembly", etc.

    --robots: Robot(s) with which to perform the task. Can be any in
        {"Panda", "Sawyer", "IIWA", "Jaco", "Kinova3", "UR5e", "Baxter"}. Note that the environments include sanity
        checks, such that a "TwoArm..." environment will only accept either a 2-tuple of robot names or a single
        bimanual robot name, according to the specified configuration (see below), and all other environments will
        only accept a single single-armed robot name

    --config: Exclusively applicable and only should be specified for "TwoArm..." environments. Specifies the robot
        configuration desired for the task. Options are {"bimanual", "single-arm-parallel", and "single-arm-opposed"}

            -"bimanual": Sets up the environment for a single bimanual robot. Expects a single bimanual robot name to
                be specified in the --robots argument

            -"single-arm-parallel": Sets up the environment such that two single-armed robots are stationed next to
                each other facing the same direction. Expects a 2-tuple of single-armed robot names to be specified
                in the --robots argument.

            -"single-arm-opposed": Sets up the environment such that two single-armed robots are stationed opposed from
                each other, facing each other from opposite directions. Expects a 2-tuple of single-armed robot names
[... 153 lines cut ...]
            # Fill out the rest of the action space if necessary
            rem_action_dim = env.action_dim - action.size
            if rem_action_dim > 0:
                # Initialize remaining action space
                rem_action = np.zeros(rem_action_dim)
                # This is a multi-arm setting, choose which arm to control and fill the rest with zeros
                if args.arm == "right":
                    action = np.concatenate([action, rem_action])
                elif args.arm == "left":
                    action = np.concatenate([rem_action, action])
                else:
                    # Only right and left arms supported
                    print(
                        "Error: Unsupported arm specified -- "
                        "must be either 'right' or 'left'! Got: {}".format(args.arm)
                    )
            elif rem_action_dim < 0:
                # We're in an environment with no gripper action space, so trim the action space to be the action dim
                action = action[: env.action_dim]

            # Step through the simulation and render
            obs, reward, done, info = env.step(action)
            env.render()

```

zhuyifengzju, 2024-03-07:
I don't know much about the device you are using, so you need to tune the conversion of the scaling yourself. There's little information about getting stuck in `input2action`. So can you set some breakpoints in https://github.com/ARISE-Initiative/robosuite/blob/48c1b8a6c077d04399a00db05694d7f9f876ffc9/robosuite/utils/input_utils.py#L181 and see which line is causing the issue?

Jericho-Ma, 2024-03-09:
已经解决，原因是device类中实时获取数据的函数未加延时。推测机理：threading使用这个函数时导致占用过高，影响后续仿真渲染等等。
‘’‘
    def run(self):
        while True:
            self.x,self.y,self.z,self.roll,self.pitch,self.yaw = omega.get_pos_and_orideg()
            self._control = [
                self.x,
                self.y,
                self.z,
                self.roll,
                self.pitch,
                self.yaw,
            ]
            #time.sleep(0.00025) 延时必须要添加
        # 夹爪功能待完善
’‘’

closed 2024-03-09 by zhuyifengzju