ARISE-Initiative/robosuite #459: using robosuite without opengl on linux server

nutsintheshell, 2024-03-07:
I'm using a linux server without opengl. when I run`import robosuite`, the following error occurs:

> Traceback (most recent call last):
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/osmesa.py", line 22, in GL

    return ctypesloader.loadLibrary(
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/ctypesloader.py", line 45, in loadLibrary
    return dllType( name, mode )
  File "/root/miniconda3/envs/tapnet/lib/python3.10/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: ('OSMesa: cannot open shared object file: No such file or directory', 'OSMesa', None)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/robosuite/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/robosuite/environments/base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/robosuite/utils/binding_utils.py", line 12, in <module>
    import mujoco
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/mujoco/__init__.py", line 47, in <module>
    from mujoco.gl_context import *
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/mujoco/gl_context.py", line 38, in <module>
    from mujoco.osmesa import GLContext as _GLContext
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/mujoco/osmesa/__init__.py", line 31, in <module>
    from OpenGL import GL
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/GL/__init__.py", line 3, in <module>
    from OpenGL import error as _error
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/error.py", line 12, in <module>
    from OpenGL import platform, _configflags
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/__init__.py", line 35, in <module>
    _load()
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/__init__.py", line 32, in _load
    plugin.install(globals())
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/baseplatform.py", line 92, in install
    namespace[ name ] = getattr(self,name,None)
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/baseplatform.py", line 14, in __get__
    value = self.fget( obj )
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/osmesa.py", line 66, in GetCurrentContext
    function = self.OSMesa.OSMesaGetCurrentContext
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/baseplatform.py", line 14, in __get__
    value = self.fget( obj )
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/osmesa.py", line 60, in OSMesa
    def OSMesa( self ): return self.GL
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/baseplatform.py", line 14, in __get__
    value = self.fget( obj )
  File "/root/miniconda3/envs/tapnet/lib/python3.10/site-packages/OpenGL/platform/osmesa.py", line 28, in GL
    raise ImportError("Unable to load OpenGL library", *err.args)
ImportError: ('Unable to load OpenGL library', 'OSMesa: cannot open shared object file: No such file or directory', 'OSMesa', None)

I wonder if there is a way to solve the problem or if I can run robosuite without opengl(I don't need to visualize any image on server). thanks

zhuyifengzju, 2024-03-07:
You would need to install one of the GL packages (egl, osmesa, glx) for mujoco to run properly.

kevin-thankyou-lin, 2024-09-07:
Osmesa installation: https://github.com/larskanis/osmesa/blob/master/README.md

closed 2024-09-07 by kevin-thankyou-lin