ARISE-Initiative/robosuite #465: How add an impedance controller in teleoperation

Jericho-Ma, 2024-03-25:
I now have a haptic device called omega7 ( As shown below ), and I have written its device file. Now I use it for teleoperation to operate the ur5e robotic arm in the simulation. I now want to [image] and use this controller to achieve better teleoperation. Where should I start to modify it. Thank you for your generous help

Jericho-Ma, 2024-03-25:
[image]

Jericho-Ma, 2024-03-25:
[image]

kevin-thankyou-lin, 2024-09-07:
So you've written the device according to [these docs](https://robosuite.ai/docs/simulation/device.html) and want to use an impedance controller? I'd create a file on the same level as the [osc controller](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/osc.py) and inherit from the `controller` class. E.g.:

```from robosuite.controllers.base_controller import Controller

class ImpedanceController:
     ...
```

closed 2024-09-18 by kevin-thankyou-lin