ARISE-Initiative/robosuite #467: OpenGL.raw.EGL._errors.EGLError

TinyDreamer1, 2024-04-02:
Hello, I'm using robosuite 1.4 with offscreen rendering. When I run the demo_video_recording.py following [instructions](https://robosuite.ai/docs/demos.html), everything including the recorded video seems okay, but when the program exits, I get errors
`Exception ignored in: <function MjRenderContext.__del__ at 0x7f1ed31689d0>
Traceback (most recent call last):
  File ".../robosuite/utils/binding_utils.py", line 199, in __del__
  File ".../robosuite/renderers/context/egl_context.py", line 149, in free
  File ".../site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
Exception ignored in: <function EGLGLContext.__del__ at 0x7f1ed3168820>
Traceback (most recent call last):
  File ".../robosuite/renderers/context/egl_context.py", line 155, in __del__
  File ".../robosuite/renderers/context/egl_context.py", line 149, in free
  File ".../site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>`
I have verified my EGL installation using [this](https://github.com/erwincoumans/egl_example) repo, and the output is 
`Loaded EGL 1.5 after reload.
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 470.57.02
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA`
what might be wrong here?

TinyDreamer1, 2024-04-12:
> See [#370 (comment)](https://github.com/ARISE-Initiative/robosuite/issues/370#issuecomment-1362042285)

Thanks, it solved my problem

closed 2024-04-12 by TinyDreamer1

cq20020226, 2024-11-04:
Sorry, could you please provide a more detailed description of where env. close() should be placed in the file?