ARISE-Initiative/robosuite #469: AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con' on local Ubuntu machine

IrvingF7, 2024-04-07:
Hi, developer of robosuite! Thank you for this project.

First of all, I am aware of these two issues 
https://github.com/ARISE-Initiative/robosuite/issues/432
https://github.com/ARISE-Initiative/robosuite/issues/424

Unfortunately, their solution doesn't seem to work in my case.

My environment is
A local Ubuntu 22.04 machine. RTX 4070 Ti with nvidia-driver-535 from the NVIDIA driver metapackage. Some simple testing, such as PyTorch and nvidia-smi, shows that the GPU and driver seem to work okay. The Ubuntu is dual-booted with a WIndows, although it lives on a separate SSD, if that matters.

To replicate this error, I run:
```pip install mujoco
pip install robosuite
python -m robosuite.demos.demo_random_action```
on a fresh Conda environment with Python 3.12.1

I set up the macro and used three 0s to select the environment and task I wanted.

The complete traceback is here:

```
dict_keys([<class 'str'>, <class 'OpenGL._opaque.EGLDeviceEXT_pointer'>, <class 'NoneType'>, <class 'tuple'>])
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/demos/demo_random_action.py", line 45, in <module>
    env = suite.make(
          ^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/base.py", line 40, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/manipulation/door.py", line 178, in __init__
    super().__init__(
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/manipulation/manipulation_env.py", line 162, in __init__
    super().__init__(
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/robot_env.py", line 214, in __init__
    super().__init__(
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/base.py", line 143, in __init__
    self._reset_internal()
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/manipulation/door.py", line 401, in _reset_internal
    super()._reset_internal()
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/robot_env.py", line 510, in _reset_internal
    super()._reset_internal()
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/base.py", line 299, in _reset_internal
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/utils/binding_utils.py", line 210, in __init__
    super().__init__(sim, offscreen=True, device_id=device_id, max_width=max_width, max_height=max_height)
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/utils/binding_utils.py", line 78, in __init__
    self.gl_ctx = GLContext(max_width=max_width, max_height=max_height, device_id=self.device_id)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/renderers/context/egl_context.py", line 129, in __init__
    EGL.eglChooseConfig(EGL_DISPLAY, EGL_ATTRIBUTES, ctypes.byref(config), config_size, num_configs)
  File "/home/irving/miniconda3/lib/python3.12/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
ctypes.ArgumentError: argument 3: TypeError: No array-type handler for type _ctypes.CArgObject (value: <cparam 'P' (0x773593bd11a0)>) registered
Exception ignored in: <function EGLGLContext.__del__ at 0x7735a14bd300>
Traceback (most recent call last):
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/renderers/context/egl_context.py", line 155, in __del__
    self.free()
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/renderers/context/egl_context.py", line 146, in free
    if self._context:
       ^^^^^^^^^^^^^
AttributeError: 'EGLGLContext' object has no attribute '_context'
Exception ignored in: <function MjRenderContext.__del__ at 0x7735a14bd440>
Traceback (most recent call last):
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/utils/binding_utils.py", line 198, in __del__
    self.con.free()
    ^^^^^^^^
AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con'```

I am not very familiar with rendering and graphics in general, but this looks like it's saying that I am starting a headless rendering? But I am running it on a local Ubuntu machine.

Interestingly, I was able to run robosuite in a similar environment on Windows by using some workarounds, despite the fact that Windows is not officially supported.

Anyway, looking forward to figure this out.

zhuyifengzju, 2024-04-08:
Hi, 
> dict_keys([<class 'str'>, <class 'OpenGL._opaque.EGLDeviceEXT_pointer'>, <class 'NoneType'>, <class 'tuple'>])
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/demos/demo_random_action.py", line 45, in <module>
    env = suite.make(
          ^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/base.py", line 40, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/manipulation/door.py", line 178, in __init__
    super().__init__(
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/manipulation/manipulation_env.py", line 162, in __init__
    super().__init__(
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/robot_env.py", line 214, in __init__
    super().__init__(
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/base.py", line 143, in __init__
    self._reset_internal()
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/manipulation/door.py", line 401, in _reset_internal
    super()._reset_internal()
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/robot_env.py", line 510, in _reset_internal
    super()._reset_internal()
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/environments/base.py", line 299, in _reset_internal
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/utils/binding_utils.py", line 210, in __init__
    super().__init__(sim, offscreen=True, device_id=device_id, max_width=max_width, max_height=max_height)
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/utils/binding_utils.py", line 78, in __init__
    self.gl_ctx = GLContext(max_width=max_width, max_height=max_height, device_id=self.device_id)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/irving/miniconda3/lib/python3.12/site-packages/robosuite/renderers/context/egl_context.py", line 129, in __init__
    EGL.eglChooseConfig(EGL_DISPLAY, EGL_ATTRIBUTES, ctypes.byref(config), config_size, num_configs)
  File "/home/irving/miniconda3/lib/python3.12/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
ctypes.ArgumentError: argument 3: TypeError: No array-type handler for type _ctypes.CArgObject (value: <cparam 'P' (0x773593bd11a0)>) registered
Exception ignored in: <function EGLGLContext.__del__ at 0x7735a14bd300>

This part is where the error comes from. It seems like the issue really comes from OpenGL. Can you double check first that if you can run mujoco rendernig at all? Maybe it is due to incorrect installation of EGL, but more information is needed.

IrvingF7, 2024-04-09:
Hi @zhuyifengzju 
Thank you so much for your reply. 

Sorry, but I forgot to mention that I tested my Mujoco rendering with 
```
python -m mujoco.viewer --mjcf=/home/{my_username}/Desktop/hello.xml
```

And I was able to see this:
[image]

zhuyifengzju, 2024-04-20:
Have you tried with running the robosuite scripts with `MUJOCO_GL="glx"` ?

michaelmunje, 2024-04-23:
Also had this issue. That line saved me a lot of debugging, thank you @zhuyifengzju  :pray:.

Charlie0257, 2024-06-22:
Hi, @zhuyifengzju 
When I try `MUJOCO_GL="glx" python xxx.py`, the errors are——
```
/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/glfw/__init__.py:914: GLFWError: (65544) b'X
11: The DISPLAY environment variable is missing'                                                            
  warnings.warn(message, GLFWError)                                                                         
/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/glfw/__init__.py:914: GLFWError: (65537) b'T
he GLFW library is not initialized'                                                                         
  warnings.warn(message, GLFWError)                                                                         
Process Process-13:                                                                                         
Traceback (most recent call last):                                                                          
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/multiprocessing/process.py", line 315, in _bootstr
ap                                                                                                          
    self.run()                                                                                              
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/multiprocessing/process.py", line 108, in run     
    self._target(*self._args, **self._kwargs)                                                               
  File "/data2/zhanghao/Lotus/lotus/libero/envs/venv.py", line 231, in _worker                              
    env = env_fn_wrapper.data()                                                                             
  File "/data2/zhanghao/Lotus/lotus/lifelong/metric.py", line 93, in <lambda>                               
    [lambda: OffScreenRenderEnv(**env_args) for _ in range(env_num)]                                        
  File "/data2/zhanghao/Lotus/lotus/libero/envs/env_wrapper.py", line 161, in __init__                      
    super().__init__(**kwargs)                                                                              
  File "/data2/zhanghao/Lotus/lotus/libero/envs/env_wrapper.py", line 56, in __init__                       
    self.env = TASK_MAPPING[self.problem_name](                                                             
  File "/data2/zhanghao/Lotus/lotus/libero/envs/problems/libero_floor_manipulation.py", line 37, in __init__
    super().__init__(bddl_file_name, *args, **kwargs)                                                       
  File "/data2/zhanghao/Lotus/lotus/libero/envs/bddl_base_domain.py", line 135, in __init__                 
    super().__init__(                                                                                       
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/manipulation/
manipulation_env.py", line 162, in __init__                                                                 
    super().__init__(                                                                                       
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/robot_env.py"
, line 214, in __init__                                                                                     
    super().__init__(                                                                                       
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/base.py", lin
e 143, in __init__                                                                                          
    self._reset_internal()                                                                                  
  File "/data2/zhanghao/Lotus/lotus/libero/envs/bddl_base_domain.py", line 735, in _reset_internal          
    super()._reset_internal()                                                                               
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/robot_env.py"
, line 510, in _reset_internal                                                                              
    super()._reset_internal()                                                                               
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/base.py", lin
e 299, in _reset_internal                                                                                   
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)                
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/utils/binding_utils.py", l
ine 210, in __init__                                                                                        
    super().__init__(sim, offscreen=True, device_id=device_id, max_width=max_width, max_height=max_height)  
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/utils/binding_utils.py", l
ine 109, in __init__                                                                                        
    self._set_mujoco_context_and_buffers()                                                                  
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/utils/binding_utils.py", l
[... cut ...]
Traceback (most recent call last):                                                                                                                                            
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/utils/binding_utils.py", line 198, in __del__                                               
    self.con.free()                                                                                                                                                           
AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con'
```

Could you give me some advice?
Thanks for any help!:)
Best,
Charlie

wkwan7, 2024-06-22:
maybe you can try the following:
`sudo apt-get update`
`sudo apt install libgl1-mesa-dev libgl1-mesa-glx libglew-dev libosmesa6-dev libgl1-mesa-glx libglfw3 libglfw3-dev`

zhuyifengzju, 2024-06-22:
Hi,

In the first example, the error was because you do not have the display set up. It's most likely that you are not running it locally with GUI. "glx" is only available when you run the code with GUI.

As for the second one, is this the full error message? It seems like the error is only about the egl not closed properly, but it is probably not the main reason why the program crashes. Please double-check if egl is installed and see if there are other error messages.

Charlie0257, 2024-06-22:
@zhuyifengzju Thanks for your suggestions!

I want to run programs based on robosuite on the server with offscreen mode.

In the second case, the completed info are——

```
Traceback (most recent call last):                                                                                                                   [431/1156]
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap                                                 
    self.run()                                                                                                                                                 
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/multiprocessing/process.py", line 108, in run                                                        
    self._target(*self._args, **self._kwargs)                                                                                                                  
  File "/data2/zhanghao/Lotus/lotus/libero/envs/venv.py", line 231, in _worker                                                                                 
    env = env_fn_wrapper.data()                                                                                                                                
  File "/data2/zhanghao/Lotus/lotus/lifelong/metric.py", line 93, in <lambda>                                                                                  
    [lambda: OffScreenRenderEnv(**env_args) for _ in range(env_num)]                                                                                           
  File "/data2/zhanghao/Lotus/lotus/libero/envs/env_wrapper.py", line 161, in __init__                                                                         
    super().__init__(**kwargs)                                                                                                                                 
  File "/data2/zhanghao/Lotus/lotus/libero/envs/env_wrapper.py", line 56, in __init__                                                                          
    self.env = TASK_MAPPING[self.problem_name](                                                                                                                
  File "/data2/zhanghao/Lotus/lotus/libero/envs/problems/libero_floor_manipulation.py", line 37, in __init__                                                   
    super().__init__(bddl_file_name, *args, **kwargs)                                                                                                          
  File "/data2/zhanghao/Lotus/lotus/libero/envs/bddl_base_domain.py", line 135, in __init__
    super().__init__(
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/manipulation/manipulation_env.py", line 162, in __init__
    super().__init__(
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/robot_env.py", line 214, in __init__
    super().__init__(
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/base.py", line 143, in __init__
    self._reset_internal()
  File "/data2/zhanghao/Lotus/lotus/libero/envs/bddl_base_domain.py", line 735, in _reset_internal
    super()._reset_internal()
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/robot_env.py", line 510, in _reset_internal
    super()._reset_internal()
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/environments/base.py", line 299, in _reset_internal
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/utils/binding_utils.py", line 210, in __init__
    super().__init__(sim, offscreen=True, device_id=device_id, max_width=max_width, max_height=max_height)
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/utils/binding_utils.py", line 78, in __init__
    self.gl_ctx = GLContext(max_width=max_width, max_height=max_height, device_id=self.device_id)
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/renderers/context/egl_context.py", line 136, in __init__
    self._context = EGL.eglCreateContext(EGL_DISPLAY, config, EGL.EGL_NO_CONTEXT, None)
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
 File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/OpenGL/error.py", line 230, in glCheckError                                  [394/1156]
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
        err = EGL_BAD_ALLOC,
        baseOperation = eglCreateContext,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x7f45b1400c40>,
                <OpenGL._opaque.EGLConfig_pointer object at 0x7f45b1400ec0>,
                <OpenGL._opaque.EGLContext_pointer object at 0x7f46a419cc40>,
                None,
        ),
        result = <OpenGL._opaque.EGLContext_pointer object at 0x7f45969661c0>
)
Exception ignored in: <function EGLGLContext.__del__ at 0x7f46a1eaab80>
Traceback (most recent call last):
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/renderers/context/egl_context.py", line 155, in __del__
    self.free()
[... cut ...] /site-packages/robosuite/renderers/context/egl_context.py", line 146, in free
    if self._context:
AttributeError: 'EGLGLContext' object has no attribute '_context'
Exception ignored in: <function MjRenderContext.__del__ at 0x7f46a1eaad30>
Traceback (most recent call last):
  File "/home/zhanghao/anaconda3/envs/lotus/lib/python3.9/site-packages/robosuite/utils/binding_utils.py", line 198, in __del__
    self.con.free()
AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con'
```

zhuyifengzju, 2024-06-22:
Have you double-checked the egl installation as I asked in previous message?

Charlie0257, 2024-06-22:
When I try `ldconfig -p | grep EGL`, the print info are——

```
libEGL_nvidia.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_nvidia.so.0
libEGL_nvidia.so.0 (libc6) => /lib/i386-linux-gnu/libEGL_nvidia.so.0
libEGL_mesa.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL_mesa.so.0
libEGL.so.1 (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so.1
libEGL.so (libc6,x86-64) => /lib/x86_64-linux-gnu/libEGL.so
```

zhuyifengzju, 2024-06-22:
There two things to try if egl is already installed
1. try to render robosuite in off-screen mode. If that works, it's probably because of the multiprocessing issue from LIBERO. Not sure how to fix the multiprocessing one, but one thing to try is to disable multiprocessing.
2. try to downgrade the version of python package OpenGL -- sometimes the version will mess up the GL.

Charlie0257, 2024-06-22:
> There two things to try if egl is already installed
>
> 1. try to render robosuite in off-screen mode. If that works, it's probably because of the multiprocessing issue from LIBERO. Not sure how to fix [...]

OK, I will test these two suggestions as soon as possible, and feedback the results.

closed 2024-09-07 by kevin-thankyou-lin

wangair, 2025-03-04:
> OK, I will test these two suggestions as soon as possible, and feedback the results.

Hi,I met the same error.Have you solved it? Thank you.

AliBehdar, 2025-04-27:
I have this problem on windows the reason was in this line 43 of .conda\envs\FenvFMP\Lib\site-packages\robosuite\utils\binding_utils.py set egl for any machine that is not macOS and solve it with changing the file 

from

if macros.MUJOCO_GPU_RENDERING and os.environ.get("MUJOCO_GL", None) not in ["osmesa", "glx"]:
    # If gpu rendering is specified in macros, then we enforce gpu
    # option for rendering
    if _SYSTEM == "Darwin":
        os.environ["MUJOCO_GL"] = "cgl"
    else:
        os.environ["MUJOCO_GL"] = "egl"
_MUJOCO_GL = os.environ.get("MUJOCO_GL", "").lower().strip()

to 

if macros.MUJOCO_GPU_RENDERING and os.environ.get("MUJOCO_GL", None) not in ["osmesa", "glx"]:
    # If gpu rendering is specified in macros, then we enforce gpu
    # option for rendering
    if _SYSTEM == "Darwin":
        os.environ["MUJOCO_GL"] = "cgl"
    else:
        os.environ["MUJOCO_GL"] = "wgl"
_MUJOCO_GL = os.environ.get("MUJOCO_GL", "").lower().strip()

Giorgio-publishing, 2025-10-08:
MUJOCO_GL="glx" xvfb-run -s "-screen 0 1280x720x24" -a python xxx.py works for me. xxx.py is the example given by official libero readme
python 3.10.18
torch 2.2.0
Ubuntu 20.04.2 LTS