ARISE-Initiative/robosuite #481: ValueError: Error: resource not found via provider or OS filesystem

bkpcoding, 2024-05-09:
Hello, when I am trying to extract data from raw.hdf5 file of robomimic, I am getting this error. I am not able to figure out what is wrong, any help appreciated!
```
input file: ../../datasets/lift/ph/demo_v141.hdf5
output file: ../../datasets/lift/ph/low_dim_v141.hdf5
 27%|███████████████████████████▎                                                                         | 54/200 [01:34<04:16,  1.76s/it]
Traceback (most recent call last):
  File "/teamspace/studios/this_studio/robomimic/robomimic/scripts/dataset_states_to_obs.py", line 425, in <module>
  File "/teamspace/studios/this_studio/robomimic/robomimic/scripts/dataset_states_to_obs.py", line 260, in dataset_states_to_obs
  File "/teamspace/studios/this_studio/robomimic/robomimic/scripts/dataset_states_to_obs.py", line 92, in extract_trajectory
  File "/teamspace/studios/this_studio/robomimic/robomimic/envs/env_robosuite.py", line 150, in reset_to
  File "/teamspace/studios/this_studio/robomimic/robomimic/envs/env_robosuite.py", line 132, in reset
  File "/teamspace/studios/this_studio/robosuite/robosuite/environments/base.py", line 252, in reset
  File "/teamspace/studios/this_studio/robosuite/robosuite/environments/base.py", line 230, in _initialize_sim
  File "/teamspace/studios/this_studio/robosuite/robosuite/utils/binding_utils.py", line 1072, in from_xml_string
ValueError: Error: resource not found via provider or OS filesystem: '/teamspace/studios/this_studio/robosuite/robosuite/models/assets/robots/panda/obj_meshes/link6_vis/link6_vis_11.obj'
```

bkpcoding, 2024-05-09:
Seems, like it was a problem with the new mujoco version. Solved it by degrading mujoco to 3.1.4.
```
pip install mujoco==3.1.4
```

closed 2024-05-09 by bkpcoding

wualbert, 2024-05-20:
I ran into the same error message `ValueError: Error: resource not found via provider or OS filesystem` when running `env.reset()` using Mujoco 3.1.5. Downgrading to Mujoco 3.1.4 also fixed the issue.

snasiriany, 2024-05-27:
To be safe, I would recommend using MuJoCo 3.1.1. 

(3.1.4 is known to have some of its own issues)