ARISE-Initiative/robosuite #486: Add an object for visual only

GreenAvocado92, 2024-05-17:
When I add an object for visual only, an error occurred like below:
```
  File "/home/idm/Shares/imitation/robosuite/robosuite/environments/manipulation/manipulation_env.py", line 162, in __init__
    super().__init__(
  File "/home/idm/Shares/imitation/robosuite/robosuite/environments/robot_env.py", line 214, in __init__
    super().__init__(
  File "/home/idm/Shares/imitation/robosuite/robosuite/environments/base.py", line 137, in __init__
    self._initialize_sim()
  File "/home/idm/Shares/imitation/robosuite/robosuite/environments/base.py", line 230, in _initialize_sim
    self.sim = MjSim.from_xml_string(xml)
  File "/home/idm/Shares/imitation/robosuite/robosuite/utils/binding_utils.py", line 1073, in from_xml_string
    model = mujoco.MjModel.from_xml_string(xml)
ValueError: Error: mass and inertia of moving bodies must be larger than mjMINVAL
Object name = cube_target_main, id = 25, line = 275, column = -1
```

zhuyifengzju, 2024-06-20:
A reference for creating a reference visual object can be found [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/objects/bread-visual.xml#L1-L15). If you are following the exact same definition, please provide more details so that we can help with.

kevin-thankyou-lin, 2024-09-07:
Closing for now

closed 2024-09-07 by kevin-thankyou-lin