ARISE-Initiative/robosuite #490: Install Error: Cannot initialize a EGL device display.

GuHuangAI, 2024-06-15:
I would like to record the issue when installing the environment and how to solve it.

GuHuangAI, 2024-06-15:
Environment:
Ubuntu22.04
RTX 4090
CUDA11.7
DRIVER:525.89.02

GuHuangAI, 2024-06-15:
After I installed the packages of requirements.txt, I ran the command `python robosuite/demos/demo_random_action.py`. However, it failed:
`ImportError: Cannot initialize a EGL device display. This likely means that your EGL driver does not support the PLATFORM_DEVICE extension, which is required for creating a headless rendering context.`

GuHuangAI, 2024-06-15:
However, it led to another error:
`libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
/home/huang/anaconda3/envs/robosuite/lib/python3.8/site-packages/glfw/__init__.py:914: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
Traceback (most recent call last):
  File "robosuite/demos/demo_random_action.py", line 45, in <module>
    env = suite.make(
  File "/home/huang/code/robosuite-master/robosuite/environments/base.py", line 40, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/huang/code/robosuite-master/robosuite/environments/manipulation/door.py", line 178, in __init__
    super().__init__(
  File "/home/huang/code/robosuite-master/robosuite/environments/manipulation/manipulation_env.py", line 162, in __init__
    super().__init__(
  File "/home/huang/code/robosuite-master/robosuite/environments/robot_env.py", line 214, in __init__
    super().__init__(
  File "/home/huang/code/robosuite-master/robosuite/environments/base.py", line 143, in __init__
    self._reset_internal()
  File "/home/huang/code/robosuite-master/robosuite/environments/manipulation/door.py", line 401, in _reset_internal
    super()._reset_internal()
  File "/home/huang/code/robosuite-master/robosuite/environments/robot_env.py", line 510, in _reset_internal
    super()._reset_internal()
  File "/home/huang/code/robosuite-master/robosuite/environments/base.py", line 299, in _reset_internal
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 211, in __init__
    super().__init__(sim, offscreen=True, device_id=device_id, max_width=max_width, max_height=max_height)
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 110, in __init__
    self._set_mujoco_context_and_buffers()
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 113, in _set_mujoco_context_and_buffers
    self.con = mujoco.MjrContext(self.model._model, mujoco.mjtFontScale.mjFONTSCALE_150)
mujoco.FatalError: gladLoadGL error
Exception ignored in: <function MjRenderContext.__del__ at 0x7f98648ffaf0>
Traceback (most recent call last):
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 199, in __del__
AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con'`

GuHuangAI, 2024-06-15:
I found this error is caused by that the file `/usr/lib/dri/swrast_dri.so` was not exist.
But in fact, there is a file named `swrast_dri.so` in the directory `/usr/lib/x86_64-linux-gnu/dri/`.
So we need to create a soft link:
`sudo mkdir /usr/lib/dri`
`cd /usr/lib/dri`
`sudo ln -s  /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so swrast_dri.so`

GuHuangAI, 2024-06-15:
Unfortunately, It was not done.
`libGL error: MESA-LOADER: failed to open swrast: /home/huang/anaconda3/envs/robosuite/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
/home/huang/anaconda3/envs/robosuite/lib/python3.8/site-packages/glfw/__init__.py:914: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
Traceback (most recent call last):
  File "robosuite/demos/demo_random_action.py", line 45, in <module>
    env = suite.make(
  File "/home/huang/code/robosuite-master/robosuite/environments/base.py", line 40, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/huang/code/robosuite-master/robosuite/environments/manipulation/door.py", line 178, in __init__
    super().__init__(
  File "/home/huang/code/robosuite-master/robosuite/environments/manipulation/manipulation_env.py", line 162, in __init__
    super().__init__(
  File "/home/huang/code/robosuite-master/robosuite/environments/robot_env.py", line 214, in __init__
    super().__init__(
  File "/home/huang/code/robosuite-master/robosuite/environments/base.py", line 143, in __init__
    self._reset_internal()
  File "/home/huang/code/robosuite-master/robosuite/environments/manipulation/door.py", line 401, in _reset_internal
    super()._reset_internal()
  File "/home/huang/code/robosuite-master/robosuite/environments/robot_env.py", line 510, in _reset_internal
    super()._reset_internal()
  File "/home/huang/code/robosuite-master/robosuite/environments/base.py", line 299, in _reset_internal
    render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 211, in __init__
    super().__init__(sim, offscreen=True, device_id=device_id, max_width=max_width, max_height=max_height)
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 110, in __init__
    self._set_mujoco_context_and_buffers()
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 113, in _set_mujoco_context_and_buffers
    self.con = mujoco.MjrContext(self.model._model, mujoco.mjtFontScale.mjFONTSCALE_150)
mujoco.FatalError: gladLoadGL error
Exception ignored in: <function MjRenderContext.__del__ at 0x7fc655100af0>
Traceback (most recent call last):
  File "/home/huang/code/robosuite-master/robosuite/utils/binding_utils.py", line 199, in __del__
AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con'`

GuHuangAI, 2024-06-15:
We found the key point: `version 'GLIBCXX_3.4.30' not found`.
To solve it, we would run the command `conda install -c conda-forge gcc`.
Finally done.
Hope can help anyone.

closed 2024-09-07 by kevin-thankyou-lin

jasonjabbour, 2025-06-02:
`conda install -c conda-forge gcc` worked for me, thanks!