ARISE-Initiative/robosuite #492: About the IK controller's Inputs

super-dashuaibi, 2024-06-28:
Hello, I'm using the IK controller recently and I have a lot of questions. I wish you can help me out in your busy schedule.
The question I'm most concerned about is which coordinate system the input pose increment (x_dpos, y_dpos, z_dpos, x_rot, y_rot, z_rot) is relative to.
In the document,it says,"  Control input actions are assumed to be relative to the current position / orientation of the end effector and are taken as the array (x_dpos, y_dpos, z_dpos, x_rot, y_rot, z_rot). "

But I've run it in demo_controller.py, and my conclusion is that the (x_dpos, y_dpos, z_dpos) is relative to the base coordinate system under mujoco, and the (x_rot, y_rot, z_rot) is relative to the coordinate system of the end flange of the robot arm.
I'm confused why this is relative to two coordinate systems, and why it's not relative to the real eef coordinate system of the end tool.

sarthakdas, 2024-07-17:
I agree very confusing

kevin-thankyou-lin, 2024-08-07:
I believe the authors implemented the inputs like so to make keyboard (and, spacemouse) teleop more intuitive.