ARISE-Initiative/robosuite #496: Could not find module ..\robosuite\utils\mujoco.dll

MinnaZhong, 2024-07-18:
Hey guys,

I am trying to use Robocasa, it requires installing Robosuite lib first.

When I run **python robocasa/scripts/setup_macros.py** , mujoco.dll could not be found.
[image]

There is no error when installing robosuite, but when I use **python -m robosuite.demos.demo_random_action** to verify it, it shows mujoco.dll could not be found again.
[image]
[image]

What should I do? Thanks,

Minna

abhihjoshi, 2024-07-18:
Are you using a Windows machine? If so, please refer to this solution here as it could be helpful: https://github.com/ARISE-Initiative/robosuite/issues/323#issuecomment-1707864112

MinnaZhong, 2024-07-23:
Hi, I am using Win10 and python3.9, I can't find mujoco folder under my env project, would you please share the mujoco.dll with me?

abhihjoshi, 2024-07-23:
Have you pip installed mujoco? I would recommend running the command `pip install mujoco=3.1.1`. The mujoco.dll file should be present after you do this.

closed 2024-08-21 by abhihjoshi

briliantnugraha, 2025-03-13:
> Have you pip installed mujoco? I would recommend running the command `pip install mujoco=3.1.1`. The mujoco.dll file should be present after you do this.

Unfortunately, for me, doing pip install alone won't copy the .dll file to robosuite folder. To solve this issue, I copy the mujoco.dll from ```mujoco/``` folder in **site-packages** to ```robosuite/utils/```. Hope it helps others, cheers~