ARISE-Initiative/robosuite #503: Clarification on Gripper Q Velocity / Q Position

Victeldo, 2024-08-02:
Hi,

In the observation space of a robosuite, several attributes of the robot are accounted for (like joint positions, joint velocities, end effector position, and end effector velocity). However, there is also a 6D "gripper_qpos" and a 6D "gripper_qvel". What exactly do these things represent. As far as I understand, the gripper position and velocity would be one dimension each (based on the aperture size and the velocity of opening/closing respectively).

Thanks,
Prithvi

kevin-thankyou-lin, 2024-08-07:
q typically refers to a quantity in a robot's joint configuration; qpos is joint position and qvel is joint velocity.

Which gripper are you looking at? Some grippers (e.g. dexterous hands) might have (or be modeled as having) multiple joints, hence `len(qpos) > 1`.

Victeldo, 2024-08-07:
I'm looking at the Robot IQ 2F-85 Gripper

Victeldo, 2024-08-07:
I'm looking at the Robot IQ 2F-85 Gripper.

kevin-thankyou-lin, 2024-08-07:
Right, that gripper likely has 6 joints if you look at the xml file - you could try to modify the xml file to replicate mimic joint like behavior (e.g. something like what's described in this README: https://github.com/berkeleyopenarms/blue_mujoco_v1)

Victeldo, 2024-08-09:
Just to follow up,

Is it possible to just switch to the 1DoF variant of the gripper as per https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/grippers/robotiq_85_gripper.py ? If so, would the robot proprioception observation space automatically update to reflect that there is only 1 DoF for the gripper (instead of a 6D array for the position / velocity respectively)?

If it isn't as simple -- if I understand correctly, I'd make the other joints mimic the returned joint value. However, when passing back the real observation space array of robot proprioception, would the other gripper joint values be automatically calculated/populated (given that I'm only passing in the qpos/qvel of one gripper joint)?

kevin-thankyou-lin, 2024-08-10:
I'm not whether the observation space updates automatically --- I'd recommend trying out these fixes and checking if the `obs` in `obs = env.step(action)` gives you the desired dimensionality for the gripper dof.

kevin-thankyou-lin, 2024-08-29:
Closing this issue for now

closed 2024-08-29 by kevin-thankyou-lin