ARISE-Initiative/robosuite #504: About compatibility and installation

robokang, 2024-08-06:
The last official update was in 2022, so it seems to have compatibility issues with the latest version, mujoco 3.2, and there are also issues with the windows installation. Do you have any update plans for that?

snasiriany, 2024-08-06:
We're working on a major release at the moment which should be out within the next few weeks.

What issues are you facing with MuJoCo 3.2?

Regarding the windows installation, there's a thread here: https://github.com/ARISE-Initiative/robosuite/issues/496. Is this helpful?

If you can share some more details on the issues you are facing, we can try to help!

robokang, 2024-08-07:
Yes, the link you sent was very helpful. In that thread I also noticed that mujoco 3.2 version is not available for installation on windows and must be downgraded to 3.1.1. I hope all these issues will be resolved in the next major release. Thanks!

snasiriany, 2024-08-07:
@abhihjoshi do you have any insight on this issue? Does MuJoCo 3.2 not work for windows at the moment?

abhihjoshi, 2024-08-21:
I don't believe I ran into any issues installing MuJoCo 3.2 on Windows. Do you mind giving more details on the issue. If you are building from source however, it might take much more time to build the Python bindings.

snasiriany, 2024-08-29:
closing this thread. feel free to re-open if you have followup comments

closed 2024-08-29 by snasiriany