ARISE-Initiative/robosuite #505: Installing in editable mode produces ModuleNotFound error

ygtkorkmaz, 2024-08-08:
Hello,

I installed the simulator from the source directly following the documentation. However, when I try to import modules when I am in another directory, I get `ModuleNotFound` errors. This doesn't happen when I write / run scripts in the robosuite folder but when I try to use robosuite in another directory, it produces this error.
[image]

kevin-thankyou-lin, 2024-08-29:
Can you run `pip list | grep robosuite` and see if you see robosuite?

ygtkorkmaz, 2024-08-30:
Yes, I see robosuite with the correct editable project path. However, I still need to append it to sys.path whenever I need to use it in my projects.

Abhiram824, 2024-08-31:
If pip is correctly showing robosuite, but you are still getting a `ModuleNotFoundError` perhaps it is the case that pip is not installing in the same location as your Python interpreter.

Could you run `pip -V` and `which -a python` to check if this is the case

Also, if possible, would you mind giving more details on the setup of your environment/dependencies so that we could replicate your issue?

kevin-thankyou-lin, 2024-09-07:
Closing for now --- feel free to reopen