ARISE-Initiative/robosuite #506: Error in quick start example

saeejithnair, 2024-08-21:
Hello,

I'm using the `robocasa_v0.1` branch and I was trying to run the following script from the quick start and I'm getting the following error. 
```python
import numpy as np
import robosuite as suite

# create environment instance
env = suite.make(
    env_name="Lift", # try with other tasks like "Stack" and "Door"
    robots="Panda",  # try with other robots like "Sawyer" and "Jaco"
    has_renderer=False,
    has_offscreen_renderer=True,
    use_camera_obs=False,
)

# reset the environment
env.reset()

for i in range(1000):
    action = np.random.randn(env.robots[0].dof) # sample random action
    obs, reward, done, info = env.step(action)  # take action in the environment
    env.render()  # render on display
```

Error:
```
(robocasa) smnair@guacamole:/pub0/smnair/robotics/robocasa/graph-robotics$ python robosuite/panda_lift.py 
Traceback (most recent call last):
  File "/pub0/smnair/robotics/robocasa/graph-robotics/robosuite/panda_lift.py", line 17, in <module>
    action = np.random.randn(env.robots[0].dof) # sample random action
  File "/pub0/smnair/robotics/robocasa/robosuite/robosuite/robots/robot.py", line 546, in dof
    dof = super().dof
AttributeError: 'super' object has no attribute 'dof'
```

Would you have any suggested workarounds? Thanks for all your hard work in open-sourcing this project!

snasiriany, 2024-08-21:
just pushed out a fix. can you pull and see if it resolved the issue? feel free to reopen if not.

snasiriany, 2024-08-21:
is this from an existing script, or one that you wrote on your end?

I think the line `action = np.random.randn(env.robots[0].dof) # sample random action` needs to be changed to:
`action = np.random.randn(env.action_spec[0].shape) # sample random action`

saeejithnair, 2024-08-21:
Hi @snasiriany, thanks I'll try this change out soon. The script is from the robosuite docs though 
[image]

saeejithnair, 2024-08-21:
Hi @snasiriany, tried the fix but it didn't work.
```
(robocasa) smnair@guacamole:/pub0/smnair/robotics/robocasa$ python graph-robotics/robosuite/panda_lift_render.py 
Traceback (most recent call last):
  File "/pub0/smnair/robotics/robocasa/graph-robotics/robosuite/panda_lift_render.py", line 18, in <module>
    action = np.random.randn(env.action_spec[0].shape)
  File "numpy/random/mtrand.pyx", line 1287, in numpy.random.mtrand.RandomState.randn
  File "numpy/random/mtrand.pyx", line 1448, in numpy.random.mtrand.RandomState.standard_normal
  File "_common.pyx", line 636, in numpy.random._common.cont
TypeError: 'tuple' object cannot be interpreted as an integer
```
I tried modifying the script like this instead `action = np.random.randn(*env.action_spec[0].shape)` but then get the following error.
```
(robocasa) smnair@guacamole:/pub0/smnair/robotics/robocasa$ python graph-robotics/robosuite/panda_lift_render.py 
Traceback (most recent call last):
  File "/pub0/smnair/robotics/robocasa/graph-robotics/robosuite/panda_lift_render.py", line 24, in <module>
    obs, reward, done, info = env.step(action)  # take action in the environment
  File "/pub0/smnair/robotics/robocasa/robosuite/robosuite/environments/base.py", line 454, in step
    self._pre_action(action, policy_step)
  File "/pub0/smnair/robotics/robocasa/robosuite/robosuite/environments/robot_env.py", line 598, in _pre_action
    robot.control(robot_action, policy_step=policy_step)
  File "/pub0/smnair/robotics/robocasa/robosuite/robosuite/robots/fixed_base_robot.py", line 222, in control
    self.recent_ee_pose[arm].push(np.concatenate((controller.ee_pos, T.mat2quat(controller.ee_ori_mat))))
AttributeError: 'JointVelocityController' object has no attribute 'ee_pos'
```

everything seems to work fine on the main branch though so i think i'll stick with the stable robosuite release for now instead of robocasa. thanks!

snasiriany, 2024-09-01:
Re-opening issue so we can investigate!

Moj79a, 2024-09-04:
hello I also have a problem running this file. I received the following error :
Exception has occurred: AttributeError
module 'robosuite' has no attribute 'make'
  File "C:\Users\Mojtaba\Desktop\nhj.py", line 5, in <module>
    env = suite.make(
          ^^^^^^^^^^
AttributeError: module 'robosuite' has no attribute 'make'

snasiriany, 2024-09-05:
@Moj79a this seems like a separate issue. Can you create a new issue request?

snasiriany, 2024-09-18:
Hi @saeejithnair this latest issue we discussed should be fixed now. Please pull from `robocasa_v0.1` branch to get the latest code.

closed 2024-09-18 by snasiriany