ARISE-Initiative/robosuite #509: error after start robosuite

Moj79a, 2024-09-04:
hi. i try to run demo file. but i cant run
this error:
[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite/scripts/setup_macros.py (macros.py:55)
Traceback (most recent call last):
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\demos\demo_random_action.py", line 1, in <module>
    from robosuite.controllers import load_controller_config
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\environments\__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\environments\base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\utils\binding_utils.py", line 27, in <module>
    ctypes.WinDLL(os.path.join(os.path.dirname(__file__), "mujoco.dll"))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2544.0_x64__qbz5n2kfra8p0\Lib\ctypes\__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: Could not find module 'C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\utils\mujoco.dll' (or one of its dependencies). Try using the full path with constructor syntax.

abhihjoshi, 2024-09-04:
Could you try to follow the steps here: https://github.com/ARISE-Initiative/robosuite/blob/master/docs/installation.md#installing-on-windows. I think you might need to `pip install mujoco`. If you have already done that, go to the location where your packages are stored and copy over the `mujoco.dll` file into Robosuite.

Moj79a, 2024-09-04:
Yes, I saw this site. I also installed Mujuko.
I also copied the Mujoco.dll file in the path that gives the error. But I received another error.

[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite/scripts/setup_macros.py (macros.py:55)
Traceback (most recent call last):
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\demos\demo_random_action.py", line 1, in <module>
    from robosuite.controllers import load_controller_config
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\environments\__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\environments\base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "C:\Users\Mojtaba\Desktop\robosuite_project\robosuite\robosuite\utils\binding_utils.py", line 54, in <module>
    raise RuntimeError(f"invalid value for environment variable MUJOCO_GL: {_MUJOCO_GL}")
RuntimeError: invalid value for environment variable MUJOCO_GL: egl

abhihjoshi, 2024-09-04:
I think step 4 in the link I sent earlier should solve this issue.

Moj79a, 2024-09-04:
thank u

closed 2024-09-04 by abhihjoshi