ARISE-Initiative/robosuite #512: problem to run quick start

Moj79a, 2024-09-05:
hello I have a problem running this file.
[image]
 I received the following error :
Exception has occurred: AttributeError
module 'robosuite' has no attribute 'make'
File "C:\Users\Mojtaba\Desktop\nhj.py", line 5, in
env = suite.make(
^^^^^^^^^^
AttributeError: module 'robosuite' has no attribute 'make'

kevin-thankyou-lin, 2024-09-07:
Did you follow the steps in [intsallation](https://robosuite.ai/docs/installation.html)?

And, are you running the script inside a conda env w/ robosuite installed?

Moj79a, 2024-09-08:
yes. i follow the steps in installation.
yes

zhuyifengzju, 2024-09-09:
Can you first try to remove / uninstall and reinstall again using `pip install -e .`? It's highly recommended to install it from source if pip install doesn't work well. Also, sometimes the virtual environment might screw up things and have some fake "robosuite" pointer where nothing is really there. So it's good to reinstall everything, just to make sure any previous installation is not messing up.

kevin-thankyou-lin, 2024-09-19:
closing for now, feel free to re-open

closed 2024-09-19 by kevin-thankyou-lin