ARISE-Initiative/robosuite #513: problem to use robosuite

Moj79a, 2024-09-06:
hello When I write the following line and try to run it, the macro.py file opens and the following error appears.
from robosuite.models import MujocoWorldBase

[image]

abhihjoshi, 2024-09-07:
For the first error, do you have a private macros files created? If not, you need to create one using the scripts/setup_macros.py file. 

For the second question, could you please provide some more detail? From what I see, it seems that it is calling another file that doesn't seem to be in Robosuite. Maybe this might be related to the issue?

Moj79a, 2024-09-07:
yes. i create private macros with using the setup_macros.py

snasiriany, 2024-09-08:
we need more context to be able to better help you. please let us know as many details as possible. have you tried importing other modules / paths in robosuite? any success? the more information you can share, the better. thank you.

snasiriany, 2024-09-18:
closing due to inactivity

closed 2024-09-18 by snasiriany