ARISE-Initiative/robosuite #515: How to collect demonstrations using a script in the config of "control_delta = True"

Felixvillas, 2024-09-13:
Hi~
I'd like to collect demonstrations in the task of "Lift" using a script implemented by myself. 
Here is my specific config of environment:
```json
"expl_environment_kwargs": {
    "control_freq": 20,
    "controller": "OSC_POSE",
    "env_name": "Lift",
    "hard_reset": false,
    "horizon": 500,
    "ignore_done": false,
    "reward_scale": 1.0,
    "robots": "UR5e",
    "gripper_types": "default",
    "render_gpu_device_id": 0,
    "render_camera": "frontview",
    "camera_names": "frontview", 
    "camera_depths": true,
    "camera_heights": 720,
    "camera_widths": 1280,
    "reward_shaping": true,
    "has_renderer": false,
    "use_object_obs": true
  },
```

Here is the core of the script in the config of "control_delta = False":
```python
from robosuite.utils import transform_utils as TU
def rotation_matrix(rot_angle, axis = 'z'):
        if axis == "x":
            return TU.quat2mat(np.array([np.cos(rot_angle / 2), np.sin(rot_angle / 2), 0, 0]))
        elif axis == "y":
            return TU.quat2mat(np.array([np.cos(rot_angle / 2), 0, np.sin(rot_angle / 2), 0]))
        elif axis == "z":
            return TU.quat2mat(np.array([np.cos(rot_angle / 2), 0, 0, np.sin(rot_angle / 2)]))

pick_pos = env.sim.data.body_xpos[env.sim.model.body_name2id(env.cube.root_body)]
final_angle = TU.quat2axisangle(TU.mat2quat(rotation_matrix(0.5*np.pi, axis="x")@rotation_matrix(0, axis="y")@rotation_matrix(0, axis='z')))
reaching_steps = 20
# 1.reach
action = np.zeros(7)
action[:3] = pick_pos
action[3:6] = final_angle
action[2] += 0.1

for i in range(reaching_steps):
        obs, reward, done, info = env.step(action)  # take action in the environment
        next_state = obs
        curr_state = next_state
        
# 2.picking
...
# 3.picking gripper
...
# 4.lifting
...
```
For simplicity, I only present the 'reach' section in the code above. As the 'control_delta = False', so when I pass an action with absolute value, it works. 

Now I'd like to collect demos in the config of "control_delta = False", so I need to pass an action with relative values. However, I am not clear whether the relative value here refers to the relative to the current states or to the initial state, so I have tried both ways.

```python
### relative to the current states
def curr_action(obs, env, gripper_state):
    current_action = np.zeros(7)
    current_action[0:3] = obs["robot0_eef_pos"]
    current_action[3:6] = TU.quat2axisangle(TU.mat2quat(env.robots[0].controller.ee_ori_mat))
    current_action[6] = gripper_state
    return deepcopy(current_action)
    
# reaching
action = np.zeros(7)
action[:3] = pick_pos
action[3:6] = final_angle
action[2] += 0.1
current_action = curr_action(curr_state, env, 0)
for i in range(reaching_steps):
    act_action = (action - current_action) / reaching_steps
    obs, reward, done, info = env.step(act_action)  # take action in the environment
    next_state = obs
    curr_state = next_state
```

```python
### relative to the initial states
def initial_action(obs, env, gripper_state):
    current_action = np.zeros(7)
    current_action[0:3] = obs["robot0_eef_pos"]
    current_action[3:6] = TU.quat2axisangle(TU.mat2quat(env.robots[0].controller.ee_ori_mat))
    current_action[6] = 0
    return deepcopy(current_action)
    
# reaching
action = np.zeros(7)
action[:3] = pick_pos
action[3:6] = final_angle
action[2] += 0.1
init_action = initial_action(curr_state, env, 0)
for i in range(reaching_steps):
    act_action = action - init_action
    obs, reward, done, info = env.step(act_action)  # take action in the environment
    next_state = obs
    curr_state = next_state

```
However, both attempts failed to collect good data under the condition "control_delta=True" (here, the gripper could not correctly approach the cube. The gripper either moved very slowly over a short distance or the robotic arm would rotate with the gripper as the center of rotation).

I wonder where the problem lies?

Thanks~

Felixvillas, 2024-09-14:
@yukezhu @cremebrule 
Sorry to bother. May I ask you two to clarify my doubts?

amandlek, 2024-09-14:
It's relative to the current end effector pose. However, the OSC controller uses a specific mujoco site as the reference. The name of the site can be found in `env.robots[0].controller.eef_name`, and then you can query its position and rotation with `obj_id = env.sim.model.site_name2id(env.robots[0].controller.eef_name)` and then `obj_pos = np.array(env.sim.data.site_xpos[obj_id])` and `obj_rot = np.array(env.sim.data.site_xmat[obj_id].reshape(3, 3))`.

Felixvillas, 2024-09-18:
Reference this issue comment: https://github.com/ARISE-Initiative/robosuite/issues/400#issuecomment-2276790526.

I have modified (perhaps not an improvement, I am not sure) the code in this issue comment as following for data collection. 

```python
import os
import sys
import joblib
import datetime
import argparse
# sys.path.append('/workspace/S/heguanhua2/robot_rl/robosuite_jimu')
import pdb

import numpy as np
import robosuite as suite
import robosuite.macros as macros
from robosuite.controllers import load_controller_config
from robosuite.utils import camera_utils as CU
from robosuite.utils import transform_utils as TU
from PIL import Image
import cv2
import random
import pickle as pkl
import json
from copy import deepcopy
import imageio

# if only use colors, set "opencv"
macros.IMAGE_CONVENTION = "opencv"

RED = "\033[31m"
GREEN = "\033[32m"
RESET = "\033[0m"

from OpenGL import GL
def ignore_gl_errors(*args, **kwargs):
    pass
GL.glCheckError = ignore_gl_errors

def imgs2video(imgs, video_dir, fps=20):
    assert len(imgs) != 0
    frame = imgs[0]
    h, w, l = frame.shape
    video = cv2.VideoWriter(video_dir, cv2.VideoWriter_fourcc(*'mp4v'), fps, (w, h))
    for img in imgs:
        video.write(img)
    video.release()
    
# def save_video(video_path_name, datas, fps=20):
#     if not video_path_name.endswith(".mp4"):
#         video_path_name += ".mp4"
    
#     video_writer = imageio.get_writer(video_path_name, fps=fps)
#     # print(f"number of eps: {len(datas)}")
#     for data in datas:
#         for img in data:
#             video_writer.append_data(img[::-1])
            
#     video_writer.close()
    
imgs = []
demo_lists = []

# def osc_control_to(env, ref_state, desc=20, threshold=0.02, info=""):
#     """
#     Works for most robots
#     Args
#     ----
#     ref_state: np.array([x, y, z, ax, ay, az, grip])
#     desc: Number of waypoints to goal
[... 293 lines cut ...]
#         imgs=imgs,
#         video_dir=video_dir,
#         fps=CONTROL_FREQ
#     )
#     sys.exit(0)
    
# video_dir = "./demo_gen/osc_control_video"
# os.makedirs(video_dir, exist_ok=True)
# video_dir = os.path.join(video_dir, "demo.mp4")
    
# imgs2video(
#     imgs=imgs,
#     video_dir=video_dir,
#     fps=CONTROL_FREQ
# )

print(f"Eps {eps}, total_eps: {total_eps}, Eps success rate: {eps / total_eps}")
```
I collected a sufficient number of correct demonstrations (at least 1 million transitions), and then trained offline with [tianshou's TD3_BC](https://github.com/thu-ml/tianshou/blob/master/examples/offline/d4rl_td3_bc.py) (this algorithm performs very well on d4rl's halfCheetah), but the results look poor. The visualization is as follows:

https://github.com/user-attachments/assets/1f50f3a1-5a27-4ac0-9fd8-b81959093403

https://github.com/user-attachments/assets/6952b9dc-3de0-46a5-8b14-b0e7b54552a1

I don't know where the problem lies, and I guess it might be because the actions cannot be accurately simulated under control_delta=True?

kevin-thankyou-lin, 2024-11-06:
Hey @Felixvillas , I believe you should be able to collect demos using (the now-renamed) `input_type=delta` by running:

`python robosuite/scripts/collect_human_demonstrations.py --robots UR5e`

kevin-thankyou-lin, 2024-11-07:
Closing for now - feel free to re-open

closed 2024-11-07 by kevin-thankyou-lin