ARISE-Initiative/robosuite #516: Translate the point clouds from different camera views to a unified world coordinate system.

pumfish, 2024-09-19:
Hi,
I'm tring to collect points cloud data from robosutie. I have successfully generated point clouds from ```rgb``` and ```depth``` data of a single camera view, but I am encountering issues when combining point clouds from different camera views.
For example, I tried to conbine ```frontview``` and ```robot0_eye_in_hand```
The result of directly combining the point clouds from both views is as follows:
[image]
After transforming the coordinate systems of both point clouds to the world coordinate system, the result is as follows:
[image]
The two point clouds do not correctly overlap.
I checked similiar issues #494, but did not resolve my problem.

Below is my scripts:
```python
import numpy as np
import open3d as o3d

import robosuite as suite
import robosuite.macros as macros
from robosuite.controllers import load_controller_config
# macros.IMAGE_CONVENTION = "opencv"

# cloud point
from robosuite.utils import camera_utils as CU
from robosuite.utils import transform_utils as TU

H = 128
W = 128

class CameraInfo:
    def __init__(self, width, height, fx, fy, cx, cy):
        self.width = width
        self.height = height
        self.fx = fx
        self.fy = fy
        self.cx = cx
        self.cy = cy

## Generate Cloud Point from
class CloudPoint:
    def __init__(self):
        pass

    def set_cam(self, env, width, height, camera_name):
        self.camera_name = camera_name
        intrinsic_matrix = CU.get_camera_intrinsic_matrix(
                env.sim, camera_name, height, width)
        self.cam = CameraInfo(width, height,
                              fx = intrinsic_matrix[0, 0],
                              fy = intrinsic_matrix[1, 1],
                              cx = intrinsic_matrix[0, 2],
                              cy = intrinsic_matrix[1, 2])

    def gen_points(self, obs, env, cam):
        self.set_cam(env, W, H, cam)
        colors = obs[cam + '_image']
        depths = obs[cam + '_depth']
        colors = colors / 255
        depths = CU.get_real_depth_map(env.sim, depths)[:,:,0]

        xmap, ymap = np.arange(depths.shape[1]), np.arange(depths.shape[0])
        xmap, ymap = np.meshgrid(xmap, ymap)
        points_z = depths
        points_x = (xmap - self.cam.cx) / self.cam.fx * points_z
        points_y = (ymap - self.cam.cy) / self.cam.fy * points_z

        mask = (points_z > 0) & (points_z < 5)
        points = np.stack([points_x, points_y, points_z], axis=-1)
        points = points[mask].astype(np.float32)
        colors = colors[mask].astype(np.float32)

        return points

     def pc_cam_to_pc_world(self, pc, extrinsic):
         extrinsic_inv = np.linalg.inv(extrinsic)
         R = extrinsic_inv[:3, :3]
         T = extrinsic_inv[:3, 3]
         pc = (R @ pc.T).T + T
         return pc

     def pc_world_to_pc_cam(self, pc, extrinsic):
         R = extrinsic[:3, :3]
         T = extrinsic[:3, 3]
         pc = (R @ pc.T).T + T
         return pc

# init env
controller_config = load_controller_config(default_controller="OSC_POSE")
controller_config['control_delta'] = False
controller_config['uncouple_pos_ori'] = False
env = suite.make(
    "Stack",
    "UR5e",
    controller_configs=controller_config,
    has_renderer=False,
    ignore_done=True,
    use_camera_obs=True,
    use_object_obs=False,
    camera_names=['frontview', 'robot0_eye_in_hand'],
    camera_heights=H,
    camera_widths=W,
    camera_depths=True,
    control_freq=20,
    horizon=200,
)

## get point cloud
CP = CloudPoint()
front_ex = CU.get_camera_extrinsic_matrix(env.sim, 'frontview')
hand_ex = CU.get_camera_extrinsic_matrix(env.sim, 'robot0_eye_in_hand')

# direct combine
front_points = CP.gen_points(obs, env, 'frontview')
hand_points = CP.gen_points(obs, env, 'robot0_eye_in_hand')
points = np.concatenate([front_points, hand_points], axis=0)
point_cloud = o3d.geometry.PointCloud()
point_cloud.points = o3d.utility.Vector3dVector(points)
o3d.io.write_point_cloud('views_coor.ply', point_cloud)

# transfer to world coordinate system
front_points = CP.pc_cam_to_pc_world(front_points, front_ex)
hand_points = CP.pc_cam_to_pc_world(hand_points, hand_ex)
points = np.concatenate([front_points, hand_points], axis=0)
point_cloud = o3d.geometry.PointCloud()
point_cloud.points = o3d.utility.Vector3dVector(points)
o3d.io.write_point_cloud('world_coor.ply', point_cloud)
```

kevin-thankyou-lin, 2024-09-19:
Thanks for raising this issue -- if I'm understanding your implementation process - you took inspiration from https://github.com/haonan16/Stow/blob/b3d3045a64992a190bbad9f25b14e83b45d2ae8b/perception/sample.py#L46-L120
but it didn't work correctly.

Did you try using OPENCV convention?
I see you commented out: `# macros.IMAGE_CONVENTION = "opencv"`

pumfish, 2024-09-20:
Thank you for your prompt reply. 
I see the code mentioned in your response. I initially thought my point cloud merging method is the same as that one, so it should work, but it didn’t. I will try the example code later.

Regarding ```# macros.IMAGE_CONVENTION = "opencv"```, doesn’t this parameter just affect the orientation of the coordinate system? I understand that it shouldn't affect the point cloud merging result. My previous experiments also proved that adding this line still resulted in incorrect outcomes.

Below is the merge result after add ```macros.IMAGE_CONVENTION = "opencv"```
[image]
It can be seen that the point cloud from the ```robot0_eye_in_hand``` just moved from the left side of the ```frontview``` to the right side, and it still does not overlap with the point cloud of the table.

closed 2024-09-27 by pumfish

kevin-thankyou-lin, 2024-09-27:
Did the example code that I linked work for you?