ARISE-Initiative/robosuite #526: Task customization

JojoSim0n, 2024-10-26:
Hi, Im looking for the tasks delivered with the robosuit. I also would like to know, if there are extention possibilies for new tasks?

abhihjoshi, 2024-10-26:
Afaik, I don’t believe there are extensions for making new tasks (though I believe this would be really interesting to consider!). When making new tasks, I think the structure typically follows those from the examples. 
1. _load_model - loads the arena and any objects as a part of your environment 
2. reward - if you intend to train a model with RL, this function is where you would calculate the reward given the current state of the end
3. _check_success - checks if the env is completed based on the state of objects/stuff in the scene.
4. _reset_internal - resets the simulation, this is where you might reset the poses of the objects in your environment.
5. _setup_observables - adds values as observables to the scene. This is important if you have things like objects that you want to get the position/rotations of and use that as part of your observation of the state of the environment.

There are other functions that you will probably have to implement when building an environment. Let us know if this helps or if you have any more questions!

JojoSim0n, 2024-10-26:
thanks for the quick reply. I think i can work with this. Next thing to ask would be the integration of new robots?

abhihjoshi, 2024-10-26:
Feel free to try out this branch (new version releasing soon!): https://github.com/ARISE-Initiative/robosuite/tree/v1.5.0

closed 2024-10-31 by abhihjoshi