ARISE-Initiative/robosuite #535: How can i import my own diy robot into robosuite? Specific steps until I can use make to create the environment?

pancake-w, 2024-10-31:
I am trying to import my own robot into robosuite. But after i have followed "Importing Raw Robot Manipiulator Models Into Robosuite",  i still have a lot of problems with the "suite.make()". What i can do is to fight with every bug and make a mess. Can someone teach me exactly what parts in this project I can modify to successfully create a robot that works like panda?

kevin-thankyou-lin, 2024-10-31:
Which tutorial were you following? @zhuyifengzju do we have a tutorial for importing custom robots?

pancake-w, 2024-11-05:
> Which tutorial were you following? @zhuyifengzju do we have a tutorial for importing custom robots?

I followed this document. I can't find a whole tutorial teaching me the details to import my custom robot.
https://docs.google.com/document/d/1bSUKkpjmbKqWyV5Oc7_4VL4FGKAQZx8aWm_nvlmTVmE/edit?tab=t.0

kevin-thankyou-lin, 2024-11-06:
Got it, we're now in the process of creating an updated tutorial for importing custom robots (CC @abhihjoshi); please stay tuned! In the mean time, if you've not done so yet, take a look at [robosuite-models](https://github.com/ARISE-Initiative/robosuite_models/tree/main/robosuite_models/robots/manipulators) to see if any existing robots match what you're importing.

kevin-thankyou-lin, 2024-11-06:
Something else that could be helpful for us @pancake-w is if you could share the relevant URDF files of the robot you'd like to import

chaitanya1chawla, 2024-11-20:
Hi @kevin-thankyou-lin,

Any updates on this?
We want to import a GR1 robot with inspire hands

zhuyifengzju, 2024-11-20:
We've already provided GR1 robot with inspire hands: https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/robots/manipulators/gr1_robot.py

prasuchit, 2025-05-03:
Hi, I'm trying to create my own environment, with a custom robot, end effector, gantry, shelf, etc. I'm running into a lot of issues. Is there any example/tutorial that can help me with this? Has anyone successfully gotten their custom env to load in robosuite so far?

gokceay, 2025-05-08:
> Got it, we're now in the process of creating an updated tutorial for importing custom robots (CC [@abhihjoshi](https://github.com/abhihjoshi)); please stay tuned! In the mean time, if you've not [...]

Hi , where can I find the tutorial? I would like to convert urdf format to xml format and create a environment with a default gripper and custom gripper.

Thanks in advance